istoreos/istoreos #1651: 关于docker部署应用的时候HTTPS的访问会失败的问题

duolabmeng6, 2024-06-19:
如果放服务器就正常的,如果放到软路由 偶尔成功 绝大部分时候失败 这怎么办啊

Error type: <class 'requests.exceptions.ConnectionError'>, message: HTTPSConnectionPool(host='dashscope.aliyuncs.com', port=443): Max retries exceeded with url: /api/v1/services/aigc/text-generation/generation (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fd783440a90>: Failed to establish a new connection: [Errno 111] Connection refused'))

duolabmeng6, 2024-06-19:
测试了一下跟网络模式有关如果使用 host 的模式就是正常的 如果是不指定 host模式就会出现这种情况 目测是dns的问题 请问一下怎么做?

duolabmeng6, 2024-06-19:
```
# docker network create my_network
networks:
  my_network:
    name: my_network
    external: true``` 初步测试是因为创建了自定义的网络导致的

duolabmeng6, 2024-06-19:
/etc/config/dockerd 尝试魔改dns
```
config globals 'globals'
	option log_level 'warn'
	option iptables '0'
	option data_root '/mnt/nvme0n1-4/docker2'
	option remote_endpoint '0'
	list registry_mirrors 'https://docker.1panel.live'
	list dns '223.5.5.5'```

duolabmeng6, 2024-06-19:
以上就是解决方案...希望有更好的办法~~是否可以在ui上配置

duolabmeng6, 2024-06-19:
另外我升级了docker版本https://download.docker.com/linux/static/stable/x86_64/  

wget https://download.docker.com/linux/static/stable/x86_64/docker-26.1.4.tgz

```
tar xzvf docker-26.1.4.tgz
sudo cp docker/* /usr/bin/
docker --version
```

jjm2473, 2024-06-21:
dns可以在部署容器的时候使用`--dns=172.17.0.1`指定。其他问题就不讨论了，既然你都自己安装了其他版本的docker，出啥问题都不奇怪

duolabmeng6, 2024-06-23:
> dns可以在部署容器的时候使用`--dns=172.17.0.1`指定。其他问题就不讨论了，既然你都自己安装了其他版本的docker，出啥问题都不奇怪

用的是docker compose 不是 docker run啊

closed 2024-09-20 by duolabmeng6