istoreos/istoreos #1655: R4S两个网口配置成桥接模式下，重启后MAC地址顺序发生变化

bai1828, 2024-06-19:
## 版本 
系统版本：iStoreOS 22.03.6 2024051012
硬件：R4SE 4G+32G标准版(不带独立MAC芯片)
## 场景描述：
R4S两个网口配置成桥接模式，网线也接在WAN口(按外壳标记)并在主路由上做静态(MAC)绑定，然后重启设备，重启后WAN口取不到静态绑定的IP，而将网线换到LAN就能拿到IP。

R5C也出现一样现象。
## 复现
在做完以下设置后重启，重启后观察IP是否与绑定的一致，不一致则网线换个网口在观察。
### 网口配置：
[image]
[image]

### 接口配置：
[image]
### br-lan(默认配置，没做更改):
[image]

jjm2473, 2024-06-21:
![](https://private-user-images.githubusercontent.com/74221228/341036444-4b90f8e8-9619-4500-8a18-280188eb2c1d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg5ODI2NzUsIm5iZiI6MTcxODk4MjM3NSwicGF0aCI6Ii83NDIyMTIyOC8zNDEwMzY0NDQtNGI5MGY4ZTgtOTYxOS00NTAwLThhMTgtMjgwMTg4ZWIyYzFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIxVDE1MDYxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNmMGM1NGJhZDk1NjUyNzZkNzgyZmJlYjVlN2Q2ZmQ3Mzg3NGZkOTg3YmE1ZWM0MjFjNDI1YjIwOTFlYjQzOWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DGH6y88lScXfZFPt0SPFMBceorfb-dGM8ObwRTxg2Go)

在高级设备选项里面看看MAC地址是不是跟上级路由器上看到的一致

bai1828, 2024-06-22:
我测试了一下，重启后不一致了，就像两个网口整体互换了位置一样