istoreos/istoreos #1656: 系统无法升级，重启后版本号不变

tonghuanming, 2024-06-19:
现版本iStoreOS 22.03.6 2024052410 升级0614那个固件

主机是X86的

搜索后和[升级不到最新的系统](https://github.com/istoreos/istoreos/issues/1200)这个现象一样

按照操作执行了系统便利工具里的修复

也检查了本地启动脚本里面没有 mount --make-shared /  （没有装clouddrive2）

查看/overlay/upgrade.log日志显示应该是正常的
`root@iStoreOS:~# cat /overlay/upgrade.log
2024-06-19 23:36:35
Running Version: 'iStoreOS 22.03.6 2024052410'
Sending TERM to remaining processes ...
Sending signal TERM to microsocks (12920)
Sending signal TERM to dns2socks (12928)
Sending signal TERM to ssr-switch (12958)
Sending signal TERM to sleep (12987)
Sending signal TERM to ssr-monitor (13217)
Sending signal TERM to S99adb-enablemo (13982)
Sending signal TERM to adb (14026)
Sending signal TERM to adb (14033)
Sending signal TERM to containerd-shim (17127)
Sending signal TERM to containerd-shim (17128)
Sending signal TERM to entrypoint.sh (17357)
Sending signal TERM to docker-entrypoi (17389)
Sending signal TERM to entrypoint.sh (17486)
Sending signal TERM to PM2 v5.4.0: God (17701)
Sending signal TERM to nginx (17796)
Sending signal TERM to nginx (17797)
Sending signal TERM to node /ql/static (17862)
Sending signal TERM to node /ql/static (18101)
Sending signal TERM to node /ql/static (18102)
Sending signal TERM to node /ql/static (18145)
Sending signal TERM to sleep (18221)
Sending signal TERM to crond (18231)
Sending signal TERM to bash (18239)
Sending signal TERM to v2ray (9211)
Sending KILL to remaining processes ...
Switching to ramdisk...
Performing system upgrade...
Extract boot sector from the image
Reading partition table from image...
Writing image to /dev/sdb1...
Writing image to /dev/sdb2...
Skip partition 3 >= 3 when upgrading
Writing new UUID to /dev/sdb...
Upgrading bootloader on /dev/sdb...
Upgrade completed`

但是升级后版本号依然不会变

tonghuanming, 2024-06-19:
查看base-files版本已经为 23-2024061415  
查询 cat /etc/os-release 
``` OPENWRT_RELEASE="iStoreOS 22.03.6 2024061415"```
不知道为什么版本号不会变
是luci显示的问题么

jjm2473, 2024-06-21:
你是不是修改过/etc/openwrt-release

closed 2024-06-21 by jjm2473

tonghm0808, 2024-06-22:
> 你是不是修改过/etc/openwrt-release

并没有，找不到问题，只能手动修改版本号到最新了，下个版本再看看

chengchen-sh, 2024-06-23:
我也有同样的问题
2024-05-24 19:57:43
Running Version: 'iStoreOS 22.03.6 2024051012'
Sending TERM to remaining processes ...
Sending signal TERM to S99adb-enablemo (14431)
Sending signal TERM to adb (14433)
Sending signal TERM to adb (14436)
Sending signal TERM to dockerd (14628)
Sending signal TERM to containerd (14883)
Sending signal TERM to docker-proxy (15268)
Sending signal TERM to docker-proxy (15281)
Sending signal TERM to docker-proxy (15564)
Sending signal TERM to docker-proxy (15578)
Sending signal TERM to containerd-shim (15759)
Sending signal TERM to containerd-shim (15762)
Sending signal TERM to docker-entrypoi (16089)
Sending signal TERM to sh (16141)
Sending signal TERM to PM2 v5.1.2: God (16615)
Sending signal TERM to nginx (16905)
Sending signal TERM to nginx (16907)
Sending signal TERM to nginx (16908)
Sending signal TERM to node /ql/build/ (17249)
Sending signal TERM to node /ql/build/ (17346)
Sending signal TERM to crond (17356)
Sending signal TERM to node (17768)
Sending signal TERM to node (17789)
Sending signal TERM to docker-proxy (20951)
Sending signal TERM to docker-proxy (20959)
Sending signal TERM to docker-proxy (20968)
Sending signal TERM to docker-proxy (20975)
Sending signal TERM to trojan-plus (26286)
Sending signal TERM to sleep (28368)
Sending signal TERM to sleep (28609)
Sending signal TERM to socks_auto_swit (28902)
Sending signal TERM to ipt2socks_TCP (29028)
Sending signal TERM to sing-box (29126)
Sending signal TERM to dns2socks (29213)
Sending signal TERM to chinadns-ng (29240)
Sending signal TERM to containerd-shim (29909)
Sending signal TERM to entry.sh (29946)
Sending signal TERM to cron (30060)
Sending signal TERM to tail (30075)
Sending signal TERM to docker-proxy (30169)
Sending signal TERM to docker-proxy (30178)
Sending signal TERM to containerd-shim (30224)
Sending signal TERM to iptv-checker-rs (30269)
Sending signal TERM to monitor.sh (30323)
Sending signal TERM to ffprobe (31331)
Sending signal TERM to docker-proxy (8635)
Sending signal TERM to docker-proxy (8643)
Sending signal TERM to containerd-shim (8667)
Sending signal TERM to alist (8727)
Sending signal TERM to ddnstod (9270)
Sending KILL to remaining processes ...
Sending signal KILL to dockerd (14628)
Sending signal KILL to iptv-checker-rs (30269)
Switching to ramdisk...
Performing system upgrade...
Extract boot sector from the image
Reading partition table from image...
Writing image to /dev/sda1...
Writing image to /dev/sda2...
Skip partition 3 >= 3 when upgrading
Writing new UUID to /dev/sda...
Upgrading bootloader on /dev/sda...
Upgrade completed

jjm2473, 2024-06-24:
反正已经升级成功了，至于为什么版本号没更新，肯定是因为/etc/openwrt-release被修改了，不管是内容被修改，还是权限被修改，在升级时都会保留下来。至于这个文件为什么被修改，那就鬼才知道了。

你们可以执行这个命令`rm -f /overlay/upper/etc/openwrt-release`，然后重启下。这样至少版本号不会错了，要是下次还出现这种问题，那就自己检查到底谁修改的openwrt-release。

tonghm0808, 2024-06-27:
> 反正已经升级成功了，至于为什么版本号没更新，肯定是因为/etc/openwrt-release被修改了，不管是内容被修改，还是权限被修改，在升级时都会保留下来。至于这个文件为什么被修改，那就鬼才知道了。
> 
> 你们可以执行这个命令`rm -f /overlay/upper/etc/openwrt-release`，然后重启下。这样至少版本号不会错了，要是下次还出现这种问题，那就自己检查到底谁修改的openwrt-release。

除了openwrt-release文件 ，显示欢迎界面的banner文件里面版本号也不对，这种文件不是变量形式存储，
应该还是安装时候没有覆盖 。如果是其他程序修改了版本号 ，欢迎界面应被全部篡改。
整个/etc文件夹里面有部分覆盖成功，部分没有成功。

jjm2473, 2024-06-28:
@tonghm0808 系统是用overlayfs来保存修改的文件，跟你说的覆盖相反。overlayfs的原理我也懒得解释，总之就是/etc/openwrt-release被修改了，肯定不是istoreos自己修改的，而是用户或者其他程序修改的。之后的升级程序会清理这几个版本文件，就算自己修改了也不会再影响版本判断。