istoreos/istoreos #1657: 无法正常访问主页

lff110, 2024-06-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我在下载蓝山云系统时通过多种方式重复下载了好几个版本，所以想重启设备断掉下载镜像文件，结果就一直无法访问设备首页了，一直是拒绝访问或者是访问超时，也通过动态域名访问也不成功，昨天又通过命令更新到最新版本了还是不能访问，我完全不知道是进行了什么操作导致无法访问的。软路由是旁路由模式，现在就是想在不重置的情况下可以恢复访问，因为我所有配置及内容还没备份。

### (2) 路由器型号和固件版本

A：友善r4se软路由，istoreos-22.03.6-2024061415-r4se-squashfs.img.gz

### (3) 详细日志和/或截图

A：无
图片看QQ频道链接:【为了不下载一个蓝山云镜像重启软路由以后通过任何方式访问都无】：https://pd.qq.com/s/3bwamkqcu?shareSource=5

bai1828, 2024-06-20:
可以进主路由看下是不是设备的ip变了，或者直接访问istoreos.local试试看能不能访问

lff110, 2024-06-20:
> 可以进主路由看下是不是设备的ip变了，或者直接访问istoreos.local试试看能不能访问

肯定没变，ping都可以ping通就是访问不了

lff110, 2024-06-20:
而且其他已经安装或者设置的功能都可以通过旁路由的IP地址进行访问

lff110, 2024-06-20:
刚刚试了一下，通过你说的那个网址也无法访问

Taoyhu, 2024-06-20:
我要吐血了 和你一样的情况 我点击了在线升级固件 升级结束就访问不了ip了  ssh docker容器都可以访问 就是进不去旁路由的后台
[image]
[image]

Taoyhu, 2024-06-20:
[image]

jjm2473, 2024-06-21:
是不是之前安装了nginx，然后又卸载了？nginx插件有bug，卸载之后不会自动enable uhttpd，导致重启就没web服务。要修复也简单，打开终端（也就是7681端口），执行`/etc/init.d/uhttpd enable; /etc/init.d/uhttpd start`

lff110, 2024-06-21:
这样确实解决了，但是我没安装nginx这个东西啊

jjm2473, 2024-06-26:
之前安装过也算

Taoyhu, 2024-09-01:
邮件已经收到！谢谢

lff110, 2024-09-01:
现在又出现该问题了，我试过用原来的方法进行操作但是没反应
[image]

lff110, 2024-09-01:
我就是想重启一下安装一下迅雷的插件，因为迅雷插件里提示安装的盘不能读写文件了

lff110, 2024-09-02:
现在的错误提示是这样的，但是该错误提示是在网页标签上提示的
Status: 500 Internal Server Error /usr/lib/lua/luci/template.lua:97: Failed to execute template 'sysauth'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/argon/header'. A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: attempt to index local 'boardinfo' (a nil value) stack traceback: [string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function 'render' /usr/lib/lua/luci/dispatcher.lua:699: in function 'include' [string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function 'render' /usr/lib/lua/luci/dispatcher.lua:699: in function 'include' [string "/usr/lib/lua/luci/view/sysauth.htm"]:1: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85> (tail call): ? (tail call): ? /usr/lib/lua/luci/dispatcher.lua:482: in function </usr/lib/lua/luci/dispatcher.lua:481>

lff110, 2024-09-02:
现在是在浏览器的标签栏里有错误提示，网页是空白的，提示为
Status: 500 Internal Server Error /usr/lib/lua/luci/template.lua:97: Failed to execute template 'sysauth'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/argon/header'. A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: attempt to index local 'boardinfo' (a nil value) stack traceback: [string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function 'render' /usr/lib/lua/luci/dispatcher.lua:699: in function 'include' [string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function 'render' /usr/lib/lua/luci/dispatcher.lua:699: in function 'include' [string "/usr/lib/lua/luci/view/sysauth.htm"]:1: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85> (tail call): ? (tail call): ? /usr/lib/lua/luci/dispatcher.lua:482: in function </usr/lib/lua/luci/dispatcher.lua:481>

closed 2024-09-02 by lff110

Taoyhu, 2025-10-08:
邮件已经收到！谢谢