istoreos/istoreos #1664: 如何在istoreos里安装homeproxy？

virtually321, 2024-06-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

bcseputetto, 2024-06-23:
尽是看了某油管博主的视频跑来的。首先不要再尝试往 iStoreOS上装 `homeproxy` 了。
如果要用 `sing-box` 请用 `passwall` ， `passwall`也支持 `sing-box`
`homeproxy`只支持 `firewall4`，而`passwall`既支持`firewall3`也支持`firewall4`
目前iStoreOS 还在使用 `firewall3`，以后iStoreOS更换防火墙工具到 `firewall4`的时候再说

huahua-er, 2024-08-29:
哈哈，我掉进去了一次，一直往下点就更新到nftable。话说测试版不是有nftable的版本吗，不跟新到正式版是有什么问题吗？

ash0080, 2024-09-04:
> 尽是看了某油管博主的视频跑来的。首先不要再尝试往 iStoreOS上装 `homeproxy` 了。 如果要用 `sing-box` 请用 `passwall` ， `passwall`也支持 `sing-box` `homeproxy`只支持 `firewall4`，而`passwall`既支持`firewall3`也支持`firewall4` 目前iStoreOS 还在使用 `firewall3`，以后iStoreOS更换防火墙工具到 `firewall4`的时候再说

这话对也不对， 毕竟passwall既没测速也没控制台也没fallback，
非常远古了, 虽然能用sing-box, 那毕竟只是套用了几个协议API而已，要啥没啥

bcseputetto, 2024-09-04:
> 这话对也不对， 毕竟passwall既没测速也没控制台也没fallback， 非常远古了, 虽然能用sing-box, 那毕竟只是套用了几个协议API而已，要啥没啥
@ash0080 这个是设计思路的不同，passwall是由自己管理节点，只把会用到的节点写入配置文件，然后以类似v2ray-core或者xray-core那种形式来利用sing-box。而你的想法，是想要类似clash内核的那种方式来利用sing-box

jjm2473, 2024-09-06:
> 这话对也不对， 毕竟passwall既没测速也没控制台也没fallback，
> 非常远古了, 虽然能用sing-box, 那毕竟只是套用了几个协议API而已，要啥没啥

@ash0080 我反倒觉得passwall简单好用呢。导入个节点就能用，有特殊需求就在黑白名单添加。
其他的例如openclash这种，除非直接订阅clash配置文件，不然配置文件有得写了。

另外，passwall是基于内核的iptables和ipset来决策代理，理论上性能优于基于tun的clash之流（未测试）。

ash0080, 2024-09-07:
> > 这话对也不对， 毕竟passwall既没测速也没控制台也没fallback，
> > 非常远古了, 虽然能用sing-box, 那毕竟只是套用了几个协议API而已，要啥没啥
> 
> @ash0080 我反倒觉得passwall简单好用呢。导入个节点就能用，有特殊需求就在黑白名单添加。 其他的例如openclash这种，除非直接订阅clash配置文件，不然配置文件有得写了。
> 
> 另外，passwall是基于内核的iptables和ipset来决策代理，理论上性能优于基于tun的clash之流（未测试）。

可能是因为我从一开始就是自己写rulesets习惯了吧，没有dashboard，看不到测速有点难受，不过passwall我认为也有一个优势的， 就是HA确实能提升数倍的网速， 虽然这个东西超级古早了， 我记得十多年前就有cow+ha的神器，passwall我在R5C上跑起来大概10%~15%cpu
homeproxy也走singbox，现在只有2%， 不过现在的版本界面有点问题，再等等吧

abandon17, 2024-09-14:
解决问题了，装个immortalwrt解决问题了！

kimcaiyuan, 2024-11-08:
> 解决问题了，装个immortalwrt解决问题了！

immortalwrt可以用lucky吗?

jjm2473, 2026-02-24:
iStoreOS 24.10 已经切换到 firewall4

closed 2026-02-24 by jjm2473