istoreos/istoreos #1665: istoreos 还是不支持I226吗？

xinzhongyouhai, 2024-06-23:
反馈bug/问题模板，提建议请删除
 iStoreOS 22.03.6, 2024061415
 -------------------------------------------------------
root@iStoreOS:~# lspci|grep -i 'eth'
01:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
02:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
03:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
04:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
root@iStoreOS:~# 

网络连接老是会从1000M变成100M~

dunkeng, 2024-06-24:
我的机器就是I226，你要不考虑一下先换个网线试试？

xinzhongyouhai, 2024-06-25:
> 我的机器就是I226，你要不考虑一下先换个网线试试？

网线没问题~实际上插拔下网线有概率恢复到1000M连接~网线问题只是错觉罢了~

jjm2473, 2024-06-27:
换好点的网线，例如6类扁网线

jjm2473, 2024-06-27:
变成百兆的时候执行`ethtool eth0`看看输出什么

xinzhongyouhai, 2024-06-30:
> 变成百兆的时候执行`ethtool eth0`看看输出什么

 好的~等我再遇到变百兆的时候试验一下再来回帖

xinzhongyouhai, 2024-07-02:
root@iStoreOS:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
root@iStoreOS:~#

xinzhongyouhai, 2024-07-02:
[image]

xinzhongyouhai, 2024-07-02:
[image]

xinzhongyouhai, 2024-07-02:
[image]

jjm2473, 2024-07-08:
如果强制设置到1000M，那你要测试下稳定性，会不会丢包啥的。

你可以写到开机命令里面。

jjm2473, 2024-07-08:
重复 #1640

closed 2024-07-08 by jjm2473