istoreos/istoreos #1666: 为什么ipv6 lan口是undefined

617450941, 2024-06-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
 
A：为什么ipv6 lan口是undefined

### (2) 路由器型号和固件版本

A： iStoreOS 22.03.6 2024061415

### (3) 详细日志和/或截图

A：
[image]

[image]
[image]

617450941, 2024-06-23:
电脑客户端 也无法获取ipv6地址

138161, 2024-06-26:
2.7、DHCP服务器--IPv6 RA设置
        默认路由器：强制的
       启用SLAAC：选择
       RA标记：受管配置 其他配置
       最长 RA 间隔：空
       最短 RA 间隔：空
       RA 生命周期：空
       RA MTU：空
       RA 跳数限制：空

138161, 2024-06-26:
2.5、DHCP服务器--高级设置
     动态 DHCP：选择
     强制：不选
     IPv4 子网掩码：灰色无法更改
     DHCP 选项：空

2.6、DHCP服务器--IPv6设置
     指定的主接口：不选
     RA 服务：服务器模式
     DHCPv6 服务：服务器模式
     通告的IPv6 DNS服务器：空
      本地IPv6 DNS服务器：选择
      通告的DNS域名：空
      NDP 代理：禁用

617450941, 2024-06-26:
这是一封自动回复邮件。已经收到您的来信，我会尽快回复。

138161, 2024-06-26:
第二步，搞定lan口
2、网络--接口--lan--编辑
2.1、常规设置
     协议：静态地址
     设备：br-lan
     开机自动运行：选择
     IPv4 地址：192.168.100.1
     IPv4 子网掩码：255.255.255.0
     IPv4 网关：灰色无法更改
     IPv4 广播地址：灰色无法更改

2.2、高级设置
     强制链路：选择
     使用默认网关：选择
     使用自定义的 DNS 服务器：空
     DNS 搜索域名：空
     DNS 权重：空
     使用网关跃点：空
     覆盖 IPv4 路由表：未指定
     覆盖 IPv6 路由表：未指定
     委托 IPv6 前缀：选择
     IPv6 分配长度：64
     IPv6 前缀过滤器：-- 请选择 --
     IPv6 后缀：::1
     IPv6 优先级：空

138161, 2024-06-26:
第三步，网络--接口--全局网络选项
     IPv6 ULA 前缀：fd00::/64
     数据包引导：选择

138161, 2024-06-26:
第五步，网络--防火墙
--自定义规则
# ipv6 NAT
ip6tables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
ip6tables -A FORWARD -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT

ip6tables -I input_rule   -p udp   --sport 546 --dport 547  -j ACCEPT

#ip6tables -A input_rule   -p ipv6-nonxt -m length --length 40 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 1  -m limit --limit 100/s --limit-burst 10  -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 2  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 3  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 4  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 128  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 129  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 130  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 131  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 132  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 133  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 134  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 135  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 136  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 137  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 138  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 139  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 140  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 141  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 142  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 143  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 144  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 145  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 146  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 147  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 148  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 149  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 151  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 152  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 153  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 154  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 157  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 158  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 160  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp --icmpv6-type 161  -m limit --limit 100/s --limit-burst 10 -j ACCEPT

jjm2473, 2024-06-27:
恢复出厂设置看看

635660902, 2024-07-07:
接口是在网桥里的，接口移出网桥就会这样

635660902, 2024-07-07:
解决了，把网桥删了，接口也删了再重新创建

jjm2473, 2024-07-08:
如果一个接口属于某个网桥，那么这个接口就不能分配ip地址了，而是要在网桥上分配