istoreos/istoreos #1668: Docker Engine版本太低请求升级一下

DcMmShan, 2024-06-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
请更新一下Docker Engine的版本，Docker Engine官方版本已经到26.1.4，咱们还在20.10.22。或者有没有什么方法可以自己升级请告知谢谢。
### (1) 具体问题

A：安装immich容器时报错“can't set healthcheck.start_interval as feature require Docker Engine v25 or later”

### (2) 路由器型号和固件版本

A：x86 intel 8400，22.03.6 2024061415

### (3) 详细日志和/或截图

A：none

dunkeng, 2024-06-24:
openwrt官方包现在最新就是20.10.22，这活不是istoreos的

jjm2473, 2024-06-27:
部署容器的时候应该可以调整参数吧，healthcheck.start_interval估计也不是必须的

closed 2024-07-08 by jjm2473

dunkeng, 2024-07-08:
我发现有个地方可能有你需要的东西：
https://github.com/kenzok8/compile-package  这个项目
https://op.dllkids.xyz/packages/x86_64/dockerd_27.0.3-93_x86_64.ipk