istoreos/istoreos #1677: iStore无法识别M2接口的固态硬盘

shixin-679, 2024-06-28:
## 1.关于你要提交的问题
无法识别M2接口的固态硬盘。

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
2.1 路由器是友善R5S，这个机器预留了一个M2固态硬盘接口，我接了一个M2固态硬盘上去，在“磁盘管理中”点击重新扫描磁盘，系统日志里提示“lsblk: /dev/mmcblk22: not a block device”。
[image]

2.2 然后把这个硬盘在Windows电脑上格式化成了“EXT4”的，再次接上去，还是提示“lsblk: /dev/mmcblk22: not a block device”。

2.3 lsblk 和 fdisk -l输出如下：
[image]
[image]
usb-1和usb-2是机器2个USB接口接的2个U盘。

2.4 请教下固态硬盘是不能加还是我哪里操作问题呢？
------------------------------------------------------------------------------------------------------------------------------------
路由器型号和固件版本：
设备型号：
FriendlyElec NanoPi R5S
固件版本：
iStoreOS 22.03.6 2024061415
内核版本：
5.10.201
[image]

jjm2473, 2024-07-08:
mmcblk22肯定跟ssd无关，也不知道这个设备哪来的。恢复出厂设置看看还有没有这个错误

jjm2473, 2024-07-08:
你是不是接了usb打印机啥的，执行`lsusb -tv`看看，把结果发出来。

shixin-679, 2024-07-08:
> 你是不是接了usb打印机的，执行`lsusb -tv`看看，把结果发出来。

[image]
2个USB3.0接口接了2个U盘做存储用的
[image]

jjm2473, 2024-07-09:
修复了 https://github.com/lisaac/luci-app-diskman/pull/77

你在istore卸载重装diskman，或者稍候有更新的时候更新下就行了

closed 2024-07-09 by jjm2473

shixin-679, 2024-07-09:
> 修复了 [lisaac/luci-app-diskman#77](https://github.com/lisaac/luci-app-diskman/pull/77)
> 
> 你在istore卸载重装diskman，或者稍候有更新的时候更新下就行了

好谢谢