istoreos/istoreos #1684: sfe的加速效果可能比offload更好

bailuxianfeiying, 2024-07-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：接上一次的issue #1256 qwrt与istoreos在跑满1000m时的cpu占用对比

### (2) 路由器型号和固件版本

A：H28K iStoreOS 22.03.6 2024062810

### (3) 详细日志和/或截图

A：istoreos开启offload和数据包引导的cpu占用情况
[image]
lean的qwrt开启sfe和数据包引导的cpu占用情况
[image]

qwrt的cpu占用明显比istoreos少，可能是sfe的效果，但不确定

[image]
关闭sfe的效果，占用仍然更少

jjm2473, 2024-07-02:
你在istoreos里面开了offload和高性能FullconeNAT了没？

bailuxianfeiying, 2024-07-02:
都是开着的

jjm2473, 2024-07-02:
你在istoreos里面开着翻墙和docker来测试？那你也在qwrt上开这些功能看看。

bailuxianfeiying, 2024-07-02:
docker里倒是没有运行任何容器

bailuxianfeiying, 2024-07-02:
我在istoreos里面关闭了包括docker和翻墙等多余的服务，现在测速结果是这样的
[image]

jjm2473, 2024-07-19:
[image]

同样h28k，开启flow offload和高性能FullconeNAT情况下。DHCP连接CPU占用率还算正常。

不过PPPoE连接的话，就只能跑到600Mbps，CPU某一核会跑到100%。

我修改了下CPU调度的脚本 https://github.com/istoreos/istoreos/commit/e4baf196336b2987a984b4c394c551c4484703f8 ，基本可以跑满千兆了。你可以下载 https://github.com/istoreos/istoreos/blob/e4baf196336b2987a984b4c394c551c4484703f8/target/linux/rockchip/rk35xx/base-files/lib/board/init.sh ，覆盖掉 /lib/board/init.sh 的内容，然后重启测试。

jjm2473, 2024-07-19:
更新固件以后测试看看

bailuxianfeiying, 2024-07-19:
目前已经更新了，等明天有时间测试下

bailuxianfeiying, 2024-07-20:
[image]
[image]
[image]
感觉调度还是有点问题，始终是cpu0占用比较多，qwrt会比较平均

jjm2473, 2024-07-21:
难道你调换了wan口和lan口？

bailuxianfeiying, 2024-07-21:
确实没有，按照默认的，eth0为lan，eth1为wan

jjm2473, 2024-07-22:
你是用h28k拨号的吗？还有试试 speedtest https://www.speedtest.net/apps

jjm2473, 2024-07-22:
对了，你要关闭数据包引导，然后重启下，这样才是我配置的CPU调度

bailuxianfeiying, 2024-07-22:
数据包引导没关，我试试

bailuxianfeiying, 2024-07-22:
[image]
[image]
我这边speedtest总是抽风，有时跑不满
但是关闭了数据包引导，貌似就只调用cpu0了

bailuxianfeiying, 2024-07-22:
h28k进行pppoe拨号

bailuxianfeiying, 2024-07-22:
我先重新刷个机试试看

jjm2473, 2024-07-22:
这是我用自己搭建的PPPoE服务器测试的结果，开了Offload和高性能FullconeNAT：
[image]
[image]
[image]

理论上，压力应该主要在cpu1和cpu3，才跟我的配置相符，不知道为什么你那还是cpu0。

speedtest有些节点经常掉速，偶尔才能跑满带宽。不知道你用的泰尔网测有没有mac端？

bailuxianfeiying, 2024-07-22:
[image]
[image]
[image]
重新刷了机，现在调度应该正常了，可能是在更新时脚本没有覆盖

bailuxianfeiying, 2024-07-22:
建议下次的更新日志里面加上“关闭数据包引导”的字样，因为按照之前版本的情况，大多数人应该是打开的，否则根本跑不满千兆

jjm2473, 2024-07-22:
嗯。
arm的板子一般都特别调矫过cpu调度，不需要开数据包引导，不过之前的固件有bug（可能驱动升级以后网卡队列名称变了），h28k的eth1实际上没调成功。

x86每台设备都不一样，没办法做最优化调矫，有些可能开了数据包引导以后性能好一些。

bailuxianfeiying, 2024-07-22:
另外，泰尔网测应该只有手机的版本，我这个是在安卓子系统安装的

jjm2473, 2024-07-22:
哦，我说怎么搜不到

bailuxianfeiying, 2024-07-22:
试了好多软件，发现还是泰尔网测，全球网测，花瓣测速之类的手机软件比较准确，稳定

bailuxianfeiying, 2024-07-22:
关于/lib/board/init.sh这个脚本，在istoreos内在线更新可能会出现没能成功覆盖的情况（我的设备在重刷后脚本才是没问题的新脚本），希望注意一下

jjm2473, 2024-07-22:
> 关于/lib/board/init.sh这个脚本，在istoreos内在线更新可能会出现没能成功覆盖的情况（我的设备在重刷后脚本才是没问题的新脚本），希望注意一下

正常情况下不会的，之前可能你修改过/lib/board/init.sh，修改过的文件在升级时是保留的。