istoreos/istoreos #1689: 新版iStoreOS中无法添加WireGuard接口

money114, 2024-07-01:
## 2. 详细叙述

### (1) 具体问题

A：系统重装下了，下了最新版的iStoreOS，在安装了wireguard相关软件包后，在创建接口的时候发现没有wireguard协议

### (2) 路由器型号和固件版本

A：iStoreOS 22.03.6 2024062810 / LuCI istoreos-22.03 branch git-24.057.30647-6659faa

### (3) 详细日志和/或截图

A：
[image]
[image]

KKfile, 2024-07-02:
我也出现过，重启就好了

money114, 2024-07-02:
嗯，确实重启就好了

mingyue5826, 2025-07-04:
大佬这几个wireguard的软件包是哪里下载的呢，我这里的软件包列表不全，缺少很多东西，也没法添加wireguard接口（没有自带，换了opkg源也没有）
固件版本是最新的：iStoreOS 24.10.2 2025070410

[image]

[image]