istoreos/istoreos #1691: Tailscale版本低， 商店里是1.32.3-1 ，官方已经到了 1.68.1

lawiker, 2024-07-02:
问题如题目，用opkg update也解决不了，求大佬关注

商店里的Tailscale版本低，还停留在1.32.3-1 ，官方已经到了 1.68.1

jjm2473, 2024-07-08:
到底是tailscale还是tailscaled？

higybibafi, 2024-07-17:
@jjm2473 tailscale 是执行程序 tailscaled是他的更高权限的守护程序，升级的话 这两个同步都要升级到同一版本应该

jeei005, 2024-08-05:
UP~
Device is too old, please upgrade it to the latest version
Failed to update machine IP

Korov, 2024-08-05:
https://pkgs.tailscale.com/stable/#static 找到自己对应的架构

比如我的
wget https://pkgs.tailscale.com/stable/tailscale_1.70.0_amd64.tgz
tar zxvf tailscale_1.70.0_amd64.tgz
service tailscale stop
cp ./tailscale_1.70.0_amd64/tailscale* /usr/sbin/
service tailscale start
tailscale version

下次升级就可以直接使用如下命令升级了
tailscale update

jjm2473, 2024-08-06:
@Korov 手动安装官方版本就不知道还是否兼容现在的luci界面和命令参数

Korov, 2024-08-07:
@jjm2473 我自己用着没问题，我没用下面这两个功能
[image]

WarterVip123, 2024-08-08:
@jjm2473 兼容luci界面，很完美

jinxiao, 2024-09-23:
两个版本都很低，我也是手动替换了，没有用luci，感觉这个东西用luci有点难发挥

Jetzow527, 2024-09-25:
> https://pkgs.tailscale.com/stable/#static 找到自己对应的架构
>
> 比如我的 wget https://pkgs.tailscale.com/stable/tailscale_1.70.0_amd64.tgz tar zxvf tailscale_1.70.0_amd64.tgz service tailscale stop cp [...]

谢谢，用你的办法升级成功了。后续也可以直接用Tailscale官方的升级命令了。

ccccorbb, 2024-12-20:
重启就会掉内网公布，又得重新up一遍内网公布，不知道你们会不会这样

ccccorbb, 2024-12-23:
> > 重启就会掉内网公布，又得重新up一遍内网公布，不知道你们会不会这样
> 
> +1

我已经解决好了，1更新到最新。2指令公布路由，3全局设置也要设置。
[image]
我现在重启就不会掉内网公布了

ccyin, 2025-04-11:
> https://pkgs.tailscale.com/stable/#static 找到自己对应的架构
>
> 比如我的 wget https://pkgs.tailscale.com/stable/tailscale_1.70.0_amd64.tgz tar zxvf tailscale_1.70.0_amd64.tgz service tailscale stop cp [...]

谢谢，更新成功了。有个细节，换版本的话， 需要替换三处文本，列如

wget https://pkgs.tailscale.com/stable/tailscale_1.82.0_amd64.tgz
tar zxvf tailscale_1.82.0_amd64.tgz
service tailscale stop
cp ./tailscale_1.82.0_amd64/tailscale* /usr/sbin/
service tailscale start
tailscale version