istoreos/istoreos #1699: 树莓派3B docker安装ubuntu， kasm无法上网

petezhou2022, 2024-07-05:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：树莓派3B dockers安装了ubuntu后，kasm内部无法上网

[image]
[image]
[image]

### (2) 路由器型号和固件版本

A：树莓派3B

### (3) 详细日志和/或截图

A：

petezhou2022, 2024-07-05:
[image]

jjm2473, 2024-07-09:
我们没出树莓派3B的固件，也许你要问问固件作者是不是改了docker的网络配置。

closed 2024-07-09 by jjm2473