istoreos/istoreos #1700: 请问支持RTL8126 5G电口么

Cranberrylemom, 2024-07-05:
请问支持RTL8126 5G电口么如题

jjm2473, 2024-07-16:
安装这个驱动试试：
[kmod-r8126_5.10.201+10.013.00-1_x86_64.zip](https://github.com/user-attachments/files/16245727/kmod-r8126_5.10.201%2B10.013.00-1_x86_64.zip)

下载后解压出里面的ipk，在istore安装这个ipk。然后执行`lsmod | grep r8126`看看是不是驱动加载了，然后再看看网卡是不是识别到了。

Cranberrylemom, 2024-07-16:
> 安装这个驱动试试： [kmod-r8126_5.10.201+10.013.00-1_x86_64.zip](https://github.com/user-attachments/files/16245727/kmod-r8126_5.10.201%2B10.013.00-1_x86_64.zip)
> 
> 下载后解压出里面的ipk，在istore安装这个ipk。然后执行看看是不是驱动加载了，然后再看看网卡是不是识别到了。`lsmod | grep r8126`

佬，OK的，能识别到了，好用！！！！

Cranberrylemom, 2024-07-16:
内网测试速度跑满了，包用的！

jjm2473, 2024-07-17:
好，那下此发布固件会集成

jjm2473, 2024-07-17:
@Cranberrylemom 你测试了最新固件不安装驱动不支持rtl8126吗？我看r8125驱动应该也支持的，是不是你没在后台的杂项设置里面启用r8125驱动？

Cranberrylemom, 2024-07-17:
> @Cranberrylemom 你测试了最新固件不安装r8126驱动就不支持rtl8126吗？我看固件自带的r8125驱动应该也支持的，是不是你没在后台的杂项设置里面启用r8125驱动？

应该是我没有启动r8125驱动，因为我的是i226网卡，没想到r8125也能驱动

Cranberrylemom, 2024-07-17:
我的r8126是另外用m2转的网口

Cranberrylemom, 2024-07-17:
[image]

Cranberrylemom, 2024-07-17:
[image]

closed 2024-07-19 by jjm2473

gelinesee, 2024-07-28:
请问能装在Mac OS 吗？ 谢谢！

gelinesee, 2024-07-29:
驱动，因为我放在NAS下载固件能用，MacOS不行。谢谢你！