istoreos/istoreos #1702: docker会自动将data-root目录挂载为分区，而且每次都用的是overlays:/overlay导致无法扩展到大磁盘

jagerzhang, 2024-07-09:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：默认docker的data-root所在的分区，istoreos只分了2G，用首页的迁移向导迁移到了 /data/docker 目录后，发现 /data/docker 会自动挂载为分区，指向的磁盘是 overlayfs:/overlay 大小还是2GB。

于是，我试着将 /overlay这个分区挂载到大磁盘分区，重启后又会恢复到2GB这个分区，也就是说docker的目录迁移并没有效果，目录变了，但是空间大小没变。

最终，我在开机启动里面加了一个 mount 大磁盘分区到 /data 和 /overlay 的机制才能解决这个问题。

大佬们帮忙看看这个是使用问题还是bug？ 使用问题的话，求官方的设置方法，如何将目录真正迁移到大磁盘分区，且重启不会失效。

### (2) 路由器型号和固件版本

A： iStoreOS 22.03.6 2024062810

### (3) 详细日志和/或截图

A：目前没有网络，就没截图，如果需要，后续联网可以随时复现。

jjm2473, 2024-07-09:
肯定是使用问题啊，你应该把硬盘挂载到/mnt/data（/data也可以，但不推荐），然后将docker迁移到/mnt/data/docker就行了

jjm2473, 2024-07-09:
不要直接把硬盘根目录当作docker数据目录，docker还要自动创建很多文件夹呢

jjm2473, 2024-07-09:
硬盘挂载以后，使用docker迁移向导怎么会有这种问题？

closed 2024-07-09 by jjm2473

jagerzhang, 2024-07-11:
> 不要直接把硬盘根目录当作docker数据目录，docker还要自动创建很多文件夹呢

没有用磁盘根目录，用的是 /data，/data用了单独的大磁盘分区

jagerzhang, 2024-07-11:
> 肯定是使用问题啊，你应该把硬盘挂载到/mnt/data（/data也可以，但不推荐），然后将docker迁移到/mnt/data/docker就行了

也就是在 /mnt 下面就没这个问题吧

jagerzhang, 2024-07-11:
[image]
目前是这样挂载的

jagerzhang, 2024-07-11:
> 肯定是使用问题啊，你应该把硬盘挂载到/mnt/data（/data也可以，但不推荐），然后将docker迁移到/mnt/data/docker就行了
按照这个方法试了下，步骤如下：
1、在web上设置挂载  /大分区 到 /mnt/data
2、前台迁移docker目录到 /mnt/data/docker
3、取消我弄的挂载，包括：
/data
/data/docker

重启系统，试了下，就成下面这样了：
[image]

jagerzhang, 2024-07-11:
最终没有挂载，直接将docker指向了这个大磁盘默认挂载的分区解决了：
[image]

猜测的问题可能是，在luci web上挂载的磁盘，重启后会消失，并不会自动挂载，所以我前面手工加到 /etc/fstab 后加到开机启动mount -a 是正常的。

jjm2473, 2024-07-17:
因为你没在首页或者挂载点那边增加配置啊，你只是临时挂载，重启后肯定就按默认的规则挂载了。而且你修改/etc/fstab干嘛，这文件根本就不需要用到，这是openwrt，不是linux，不要把fstab的经验套用过来。如果你真要手动编辑配置文件，那也是/etc/config/fstab。

而且首页都有挂载功能了怎么不用，这个功能起码会保存到配置文件中。