istoreos/istoreos #1703: 23.05的源编译有问题

dd-bpir3, 2024-07-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
编译提示有问题
### (1) 具体问题

A：编译后打不开web界面

### (2) 路由器型号和固件版本

A：bpi-r3

### (3) 详细日志和/或截图

A：
WARNING: Makefile 'package/feeds/telephony/asterisk/Makefile' has a dependency on 'libjwt', which does not exist
WARNING: Makefile 'package/feeds/third/luci-app-baidupcs-web/Makefile' has a dependency on 'baidupcs-web', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6-json', which does not exist

 make[3] -C feeds/third/luci-theme-argon compile
luci-theme-argon does not support CSS source minification
luci-theme-argon does not support CSS source minification
 make[3] -C feeds/kenzo/luci-app-quickstart compile
luci-app-quickstart does not support JavaScript source minification
luci-app-quickstart does not support CSS source minification
luci-app-quickstart does not support JavaScript source minification
luci-app-quickstart does not support CSS source minification
 make[3] -C feeds/kenzo/luci-app-istorex compile
luci-app-istorex does not support JavaScript source minification
luci-app-istorex does not support CSS source minification
luci-app-istorex does not support JavaScript source minification
luci-app-istorex does not support CSS source minification

jjm2473, 2024-07-16:
没适配bpi-r3。

如果你要自己适配或编译，别忘了openwrt默认的软件包可能不包含web界面，你可以参考其他机型的istoreos固件的配置文件，例如 https://fw0.koolcenter.com/iStoreOS/rpi4/config.seed https://fw0.koolcenter.com/iStoreOS/rpi4/feeds.conf

jjm2473, 2024-07-16:
顺便提一句，23.05我们并没有维护，就发过几个x86的测试固件而已。打算以后直接上24版。建议你还是用22.03的代码。

closed 2024-07-16 by jjm2473

dd-bpir3, 2024-07-17:
>顺便提一句，23.05我们并没有维护，就发过几个x86的测试固件而已。打算以后直接上24版。建议你还是用22.03的代码。
[image]
[image]
22.03没适配bpi-r3，本人小白想用22.03源码编译，需要把23.05下target/linux/mediatek/filogic复制到22.03下吗？还是怎么操作？