istoreos/istoreos #1709: 商店里的Lucky是2.4.8现在最新的是2.10.5了求更新

moxiaobai240320, 2024-07-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：
[image]
[image]

Hopers, 2024-07-16:
可以去作者主页下载对应硬件的，不用在意有没有openwrt版本的 https://github.com/gdy666/lucky/releases
然后在lucky首页“上传新版本”更新，我一直这么更新下来，还没发现有什么问题

Korov, 2024-08-05:
[image]
自己下载对应版本升级就行了

wqinf, 2024-08-18:
> ![image](https://private-user-images.githubusercontent.com/45894441/354992796-026e4be2-2f7a-4ca1-87ec-314094a642a7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

我就是这么做的，完全没问题，楼主可以放心参考