istoreos/istoreos #1713: R5S根据官方教程TF卡刷机问题

nmapq, 2024-07-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

R5S刷入：r5s-sdcard-0610.img.gz版本后，可以成功启动，并进行配置界面，配置完毕，模拟停电重启，无法成功启动。指示灯为红色常亮
R5S根据SD卡刷写教程，如 istoreos-x.x.x-2023032001-r5s-squashfs.img.gz，均可正常亮灯，但无法打开100.1页面。

### (2) 路由器型号和固件版本

[image]
[image]

### (3) 详细日志和/或截图

因没有显示器且断电后无法启动，因而没有日志和截图

nmapq, 2024-07-19:
有emmc，但无法刷机到EMME

jjm2473, 2024-07-20:
你的emmc里面是不是刷了其他作者的固件？

nmapq, 2024-07-20:
> 你的emmc里面是不是刷了其他作者的固件？

当时emmc已经被格式化了，目前已经通过FriendlyWrt把istoreOS刷到了EMMC里面。现在遇到的问题是插入SD卡后，断电无法启动，很尴尬。

AniberMokie, 2024-08-07:
> > 你的emmc里面是不是刷了其他作者的固件？
> 
> 当时emmc已经被格式化了，目前已经通过FriendlyWrt把istoreOS刷到了EMMC里面。现在遇到的问题是插入SD卡后，断电无法启动，很尴尬。

SD卡中有其它固件是吗？”插入SD卡后，断电无法启动“是指无法启动SD卡中的固件呢？还是指R5S无法开机，EMMC的固件也进不去呢？

另外，EMMC格式化并不能删除原来EMMC里固件的uboot引导，需要写0覆盖掉原来的uboot才算删除

nmapq, 2024-08-30:
> > > 你的emmc里面是不是刷了其他作者的固件？
> > 
> > 
> > 当时emmc已经被格式化了，目前已经通过FriendlyWrt把istoreOS刷到了EMMC里面。现在遇到的问题是插入SD卡后，断电无法启动，很尴尬。
> 
> SD卡中有其它固件是吗？”插入SD卡后，断电无法启动“是指无法启动SD卡中的固件呢？还是指R5S无法开机，EMMC的固件也进不去呢？
> 
> 另外，EMMC格式化并不能删除原来EMMC里固件的uboot引导，需要写0覆盖掉原来的uboot才算删除

目前当把SD卡去掉后，是可以正常使用的，但插入SD卡就优先从SD启动系统。