istoreos/istoreos #1714: IPv6获取了PD前缀，lan下设备无法访问ipv6网站

flame-222, 2024-07-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
istoreos获取了IPv6地址和pd前缀，设备可以访问IPv6网站，但lan口下挂设备无法访问
### (1) 具体问题

A：(istoreos无任何插件)
使用路由器上终端访问6.ipw.cn时，可以正常返回ip
使用下挂无线AP访问testipv6.cn,显示无IPv6

### (2) 路由器型号和固件版本

A：x86，倍控4网N100
B:  22.03.6 2024062810

### (3) 详细日志和/或截图

A：
[image]
[image]

flame-222, 2024-07-19:
注：lan口下挂设备均已被分配到了IPv6地址

flame-222, 2024-07-20:
[image]
[image]

flame-222, 2024-07-20:
经检验，现在的情况是使用系统便利工具的“打开IPv6”后lan口下挂设备均不分配IPv6地址了

138161, 2024-07-22:
网络--lan--配置，都有PD前缀了，比我们完美太多

pcsakura, 2024-07-26:
抖音和b站都有相关的视频介绍。wan6下发pd之后，看看lan的相关设置，还有dhcp有没有取消ipv6的dns解析过滤。

leux0, 2024-08-05:
这个勾去掉了没
[image]

jagerzhang, 2025-01-04:
最后解决了吗？一模一样的环境和问题。。。

pcsakura, 2025-01-04:
> 最后解决了吗？一模一样的环境和问题。。。

太久了，都忘了怎么解决的了，b站抖音多招一下，就那几个配置。现在又不用软路由了，