istoreos/istoreos #1716: 软路由自身网速正常，但下游设备缓慢

HfSrJun, 2024-07-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
背景：家中接入千兆下行，百兆上行的宽带。由软路由拨号。软路由下接万兆交换机，万兆交换机下接千兆交换机。千兆交换机下接电脑A，电脑B，NAS。
问题：在几个月前，新的软路由启用后不久，发现部分网站（例如UU加速器更新，明日方舟的CDN）连接极其缓慢（仅有不太到1MBps），但绝大部分情况（例如steam下载）速度正常。而前几天，在更新固件以及安装“istore增强”和测速等插件后，电脑A与B的下行网速也变得极其缓慢（测速网的下载速度只有不到10MBps，steam的下载速度也在数个MB徘徊，极少到10以上），上传速度正常百兆。
首先我进行了对内网的排查，经过测试，从电脑B向NAS服务器收发文件，均可正常跑满内网的千兆带宽。在A和B电脑上分别用Homebox插件测试与软路由之间的网速，得到的结果均是下载速度为200+Mbps，上传速度为900+Mbps。两者同时跑下载速度，均能达到200+Mbps的带宽，在软路由也能观察到对应大小的出站流量（400Mbps以上）。
关键在于对软路由本身的测试。我在软路由的docker中安装了ubuntu虚拟机系统（来自插件商店），docker的网络可视作软路由本身的网络。在Ubuntu的浏览器内，测速网的网速可正常跑到入户宽带的数值，且前文“部分网站”（这里用明日方舟下载来测试）的速度也是正常的。
我应该如何开始排查问题所在？希望能得到技术支持。

### (2) 路由器型号和固件版本

A：Dell R220服务器，CPU与内存均绰绰有余。自带2个千兆网口，用于接入WAN。插入一张双口万兆网卡，以主备用方式接交换机。固件已升级为最新。

### (3) 详细日志和/或截图

A：尚未开始下一步排查，详见描述

jjm2473, 2024-07-20:
首先排除是不是wifi问题，只用有线环境测试。
测试时确保没有其他设备占用带宽。
关闭ipv6试试。

HfSrJun, 2024-07-20:
> 首先排除是不是wifi问题，只用有线环境测试。
> 测试时确保没有其他设备占用带宽。
> 关闭ipv6试试。

内网问题已经排查过了，AB电脑均是有线连接(详见排查过程的描述)。况且内网环境是一直没动过的，先前速度正常。
其他设备占用带宽这种低级错误也是不会出现的，第一时间我就查看了软路由的统计来查找是否有设备占用了带宽。
至于关闭ipv6，稍后我试验一下。难道是ubuntu系统没有使用ipv6导致的差异吗?

HfSrJun, 2024-07-21:
> 首先排除是不是wifi问题，只用有线环境测试。 测试时确保没有其他设备占用带宽。 关闭ipv6试试。

已经尝试过禁用ipv6，问题依旧

jjm2473, 2024-07-22:
istoreos是运行中虚拟机中吗？

如果怀疑固件版本的问题，可以试试前几个版本固件。

istore增强就不用管了，肯定不会影响网络，除非机器内存非常小。

HfSrJun, 2024-07-22:
> istoreos是运行在虚拟机中吗？
> 
> 如果怀疑固件版本的问题，可以试试前几个版本固件。
> 
> istore增强就不用管了，肯定不会影响网络，除非机器内存非常小。
> 

系统是直接安装到实体机上面的
现在我暂时不在家，只能远程操作家里电脑测试，所以重装系统有些难以操作，等回家后可以试试。
另外我曾经安装使用过clash，后来记不清什么原因我重置了系统，然后导入了备份的设置，但软件自然是没有了，我也没重装clash。会不会是遗留了什么代理之类的设置在?但我并没有找到可疑的设置参数。

HfSrJun, 2024-07-26:
> istoreos是运行在虚拟机中吗？
> 
> 如果怀疑固件版本的问题，可以试试前几个版本固件。
> 
> istore增强就不用管了，肯定不会影响网络，除非机器内存非常小。
> 

经过进一步的实验，发现问题实际还是和最开始一样，即:
1.软路由下接的实体机，无论是通过无线连接的笔记本电脑，手机又或者有线连接的台式机，服务器，均存在同样的网速问题
2.网速问题表现为，部分链接，例如鸣潮的更新，steam的下载，利用迅雷下载win11镜像，速度虽然未达到宽带的极限(千兆)，但也有数十MBps，暂且可以认为正常。
3.其他部分链接，例如明日方舟的下载与更新，测速网，uu加速器更新，速度均异常缓慢。缓慢程度不相同，例如测速网的结果为5MBps左右，但明日方舟下载目测只有约1-2MBps。
4.利用istore中的内网测速(homebox)插件，测试台式机(有线)到软路由的内网速度，上传正常，为千兆(网口与交换机为千兆)，但下载只有200+或300+兆。这个速度高于(3)的速度，但是低于(2)的速度。
5.利用istore的Ubuntu桌面插件，在软路由docker中安装一个带远程桌面的Ubuntu镜像。Ubuntu中测试(3)中的网速缓慢项(测速网以及明日方舟下载)均能跑满宽带上限。访问homebox网页并测速，上下行均为万兆(软路由的LAN口为万兆，接入万兆交换机后又下接千兆交换机，才接入了台式机)

另:已经排除过手机与电脑中代理软件的问题了，将clash切到直连或者关闭系统代理，问题不变

jjm2473, 2024-07-26:
抓包看看吧，是不是有ip冲突或者环路。或者断开局域网其他设备，单独测试某台机器的网速。另外，有没有试过恢复出厂设置

HfSrJun, 2024-07-26:
> 抓包看看吧，是不是有ip冲突或者环路。或者断开局域网其他设备，单独测试某台机器的网速。另外，有没有试过恢复出厂设置

局域网应该是不存在ip冲突或者环路这种低级错误的，因为网络中的所有设备都是我的设备。
现在我的想法是干脆直接还原系统(传说中的重装解决90%疑难杂症)，或者说是把局域网断掉，用设备直接连到软路由测试。但这些想法都得我在面前才好操作，近期我一直不在家，等我回家才有机会进一步研究。

Mioscout, 2024-08-10:
你动过DNS设置吗？
去dnsmaq里面多加几个转发然后基本设置选所有服务器试试

HfSrJun, 2024-08-10:
> 你动过DNS设置吗？ 去dnsmaq里面多加几个转发然后常规那选所有服务器试试

DNS有114DNS和腾讯的DNS（DDNS服务的域名是DNSPOD提供的），又添加了几个国内DNS尝试，问题依旧。
另外如果是DNS服务器问题，那么docker里面的Ubuntu理论上也应该受影响。

Mioscout, 2024-08-10:
> > 你动过DNS设置吗？ 去dnsmaq里面多加几个转发然后常规那选所有服务器试试
> 
> DNS有114DNS和腾讯的DNS（DDNS服务的域名是DNSPOD提供的），又添加了几个国内DNS尝试，问题依旧。 另外如果是DNS服务器问题，那么docker里面的Ubuntu理论上也应该受影响。

我前段时间因为卡坏了刷新卡系统也遇到了类似的问题，我自己试出来是passwall导入配置的问题，然后我恢复passwall+dnsmaq改了设置就好了。
仅供参考

HfSrJun, 2024-08-10:
> > > 你动过DNS设置吗？ 去dnsmaq里面多加几个转发然后常规那选所有服务器试试
> > 
> > 
> > DNS有114DNS和腾讯的DNS（DDNS服务的域名是DNSPOD提供的），又添加了几个国内DNS尝试，问题依旧。 另外如果是DNS服务器问题，那么docker里面的Ubuntu理论上也应该受影响。
> 
> 我前段时间因为卡坏了刷新卡系统也遇到了类似的问题，我自己试出来是passwall导入配置的问题，然后我恢复passwall+dnsmaq改了设置就好了。 仅供参考

我之前安装过openclash，也还原过系统并重新导入了配置，经过检查clash的配置文件是有所残留的。
在尝试重新安装openclash，以及清除旧有配置文件后，并没有对问题产生影响。
还是等有机会，彻底重新安装一遍系统测试一下吧

HfSrJun, 2024-09-15:
经过对固件的刷新，更换版本以及更换其他固件测试，问题应当是内网导致，经过排查应当是交换机出现了问题

closed 2024-09-15 by HfSrJun

jjm2473, 2024-09-18:
不会是一个设备有两根网线同时接一个交换机了吧？

HfSrJun, 2024-09-18:
> 不会是一个设备有两根网线同时接一个交换机了吧？

有，但正确配置了链路聚合