istoreos/istoreos #1717: ddns选择dnspod_cn时报错ERROR : [] - Terminate process

smockgithub, 2024-07-20:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

设置远程域名ddns的时候，选择了myddns_ipv4和Dnspod，并填写了Dnspod的用户名密码后。

ddns报错：update_dnspod_cn.sh

141002  note : PID '11267' started at 2024-07-20 14:10
 141002       : ddns version  : 2.8.2-29
 141002       : uci configuration:
ddns.myddns_ipv4.check_unit='minutes'
ddns.myddns_ipv4.domain='dy.wabaoge.com'
ddns.myddns_ipv4.enabled='1'
ddns.myddns_ipv4.force_unit='minutes'
ddns.myddns_ipv4.interface='wan'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.lookup_host='dy.wabaoge.com'
ddns.myddns_ipv4.password='***PW***'
ddns.myddns_ipv4.retry_unit='seconds'
ddns.myddns_ipv4.service_name='dnspod.cn'
ddns.myddns_ipv4.use_ipv6='0'
ddns.myddns_ipv4.use_syslog='2'
ddns.myddns_ipv4.username='smock130_163@163.com'
ddns.myddns_ipv4=service
 141002       : verbose mode  : 0 - run normal, NO console output
 141002       : check interval: 600 seconds
 141002       : force interval: 4320 seconds
 141002       : retry interval: 60 seconds
 141002       : retry counter : 0 times
 141002       : No old process
 141002       : last update: never
 141002       : Detect registered/public IP
 141002       : #> /usr/bin/host -t A dy.wabaoge.com  >/var/run/ddns/myddns_ipv4.dat 2>/var/run/ddns/myddns_ipv4.err
 141002       : Registered IP '47.113.54.45' detected
 141002  info : Starting main loop at 2024-07-20 14:10
 141002       : Detect local IP on 'network'
 141002       : Local IP '116.148.243.171' detected on network 'wan'
 141002       : Update needed - L: '116.148.243.171' <> R: '47.113.54.45'
 141002       : parsing script '/usr/lib/ddns/update_dnspod_cn.sh'
 141002       : #> /usr/sbin/curl -Ss --noproxy '*' -d 'login_token=dnspod账号,***PW***&format=json&domain=yuming.com&sub_domain=dy' https://dnsapi.cn/Record.List
 141002 ERROR : [] - Terminate process
 141002  WARN : PID '11267' exit WITH ERROR '1' at 2024-07-20 14:10

注：#> /usr/sbin/curl -Ss --noproxy日志后面的用户名、域名我做了替换，实际上是在我dnspod域名列表里的。
[image]

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：固件版本
iStoreOS 22.03.6 2024062810
内核版本
5.10.201

### (3) 详细日志和/或截图

[image]
[image]
[image]

138161, 2024-07-22:
腾讯的我也没搞出来，用ddns-go可以
DDNS-go
1、DnsPod腾讯云
ID：xxxx
Token：xxxxx

2、IPV4填写
通过网卡获取 eth0
域名：xxxxxxxxx 
3、IPV6填写
通过网卡获取
域名：xxxxxxx

4、Webhook设置（微信推送）
URL填写
https://sctapi.ftqq.com/SCT207983TXVw9NhJF4yBA2Lz6sjq0z7EW.send?title=公网IP变化&desp=
#{ipv4Addr}
#{ipv6Addr}

smockgithub, 2024-07-22:
我大概知道原因了，我明天试试再来反馈

smockgithub, 2024-07-23:
[image]

成功了，感谢

closed 2024-07-23 by smockgithub

138161, 2024-07-23:
啥原因？分享下，我还没搞出来过