istoreos/istoreos #1725: 云编译3399机型失败了，104-pinctrl-rockchip-rk3308bs.patch 补丁无法安装

xiaomeng9597, 2024-07-25:
编译错误日志

Applying /mnt/workdir/openwrt/target/linux/rockchip/armv8/patches-5.10/104-pinctrl-rockchip-rk3308bs.patch using plaintext: 
patching file drivers/pinctrl/pinctrl-rockchip.c
Hunk #1 FAILED at 103.
Hunk #2 succeeded at 198 (offset -111 lines).
Hunk #3 FAILED at 388.
Hunk #4 succeeded at 1461 (offset -179 lines).
Hunk #5 succeeded at 1668 (offset -179 lines).
Hunk #6 succeeded at 1731 (offset -179 lines).
Hunk #7 succeeded at 1747 (offset -179 lines).
Hunk #8 succeeded at 1845 (offset -179 lines).
Hunk #9 succeeded at 1861 (offset -179 lines).
Hunk #10 succeeded at 3722 (offset -174 lines).
Hunk #11 succeeded at 3753 (offset -174 lines).
Hunk #12 succeeded at 3963 (offset -174 lines).
2 out of 12 hunks FAILED -- saving rejects to file drivers/pinctrl/pinctrl-rockchip.c.rej
Patch failed!  Please fix /mnt/workdir/openwrt/target/linux/rockchip/armv8/patches-5.10/104-pinctrl-rockchip-rk3308bs.patch!
make[4]: *** [Makefile:33: /mnt/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.221/.prepared_55308123a6ae61b537a4899b659b2fe8] Error 1
make[4]: Leaving directory '/mnt/workdir/openwrt/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/mnt/workdir/openwrt/target/linux'
time: target/linux/compile#8.97#6.97#11.73
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/mnt/workdir/openwrt'
make[1]: *** [target/Makefile:23: /mnt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/mnt/workdir/openwrt'
make: *** [/mnt/workdir/openwrt/include/toplevel.mk:234: world] Error 2

xiaomeng9597, 2024-07-25:
错误截图
[image]

xiaomeng9597, 2024-07-25:
相关链接 https://github.com/xiaomeng9597/iStoreOS-For-RK33XX/actions/runs/10101099680/job/27933914017

jjm2473, 2024-08-02:
哦，当时正在合并openwrt 22.03.7代码，这个补丁还没更新呢。现在已经好了

closed 2024-08-02 by jjm2473