istoreos/istoreos #1726: samba共享拒绝访问

N1k1Zhang, 2024-07-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
1.在路由器上“统一文件共享”设置了共享文件夹，协议为samba和webdav，在Windows里面通过samba协议访问，提示拒绝访问，用户名密码都是正确的。
2. 手机上通过webdav协议访问可以正常访问。
3. 以前某个版本的估计是没有这个问题的，某次更新固件后，出现了如1所述的问题。

### (2) 路由器型号和固件版本

A：设备型号
Raspberry Pi 4 Model B Rev 1.4
固件版本
iStoreOS 22.03.6 2024071911

### (3) 详细日志和/或截图

A：
[image]

youli42, 2025-06-19:
也有这个问题，测试发现是使用局域网ip可以访问，但是虚拟局域网（easytier）就无法访问
一通乱调之后正常连也连不上了

CT940828071, 2025-08-20:
参考链接：https://www.bilibili.com/opus/782613801149136898

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/2122/1

hkongm, 2026-08-21:
昨天刚解完，是 UI 创建的 samba 用户没有同时创建系统级同名账户导致的，我想这是个bug，遇到的朋友应该不在少数。
而且samba默认开了 1.1 兼容，应该去掉的。