istoreos/istoreos #1729: istoreos 的docker拉取报错

jayjoke, 2024-07-28:
## 2. 详细叙述

### (1) 具体问题

拉不了镜像

### (2) 路由器型号和固件版本

x86  2024071911版本

### (3) 详细日志和/或截图

[image]

linvex, 2024-07-31:
看起来是代理有问题，换个梯子试一下

jjm2473, 2024-08-02:
istore增强启用了吗？访问其他网站是不是正常？

zyrobem, 2024-10-04:
我用了增强也不行 发现DOCKER设置那里的索引服务器还是原来的地址 不能更改

ArKurt, 2024-11-16:
咱也是一模一样的问题？

xiaomo-xxx, 2024-11-16:
> 咱也是一模一样的问题
我的应该算是解决了，大概是硬盘的问题

ArKurt, 2024-11-16:
> > 咱也是一模一样的问题
> > 我的应该算是解决了，大概是硬盘的问题

哦？请问怎么解决呢？

FDin, 2025-01-07:
笑了，这个问题大陆地区都一样，有人在dockerhub搞事情，导致GFW管理员把dockerhub拉黑了。dockerhub更硬气，反向把大陆IP也全拉黑了。
解决办法就是自己找加速放在mirrors里面就好了，不是BUG。都会用github了，找个加速地址难不倒各位吧。

jjm2473, 2025-01-07:
国内用户在istore里面安装“istore增强”插件，安装以后会自动创建一个docker的本地代理，并配置好docker使用这个代理。

King-hsw, 2025-01-24:
> 国内用户在istore里面安装“istore增强”插件，安装以后会自动创建一个docker的本地代理，并配置好docker使用这个代理。

[image]
我想请问一下，我就是安装了这个增强，但是尝试拉一个镜像还是会有问题

[image]
感觉代理没有生效