istoreos/istoreos #1731: istoreos内置APP应用问题

q994493141, 2024-07-29:
应用APP中的tailscale的版本对应更新一下呗 现在的1.39版本很老了  现在最新版是1.70的了

bcseputetto, 2024-07-30:
```

bcseputetto, 2024-07-30:
```
Fetch feed list for x86_64
Update feeds index
Downloading https://istore.linkease.com/repo/all/nas_luci/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_nas_luci
Downloading https://istore.linkease.com/repo/all/store/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_store
Downloading https://istore.linkease.com/repo/x86_64/nas/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_nas
Downloading https://istore.linkease.com/repo/all/compat/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/istore_compat
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/targets/x86/64/packages/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/openwrt_core
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/base/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/openwrt_base
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/luci/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/openwrt_luci
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/packages/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/openwrt_packages
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/routing/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/openwrt_routing
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/openwrt_telephony
Downloading https://istore.linkease.com/repo/all/meta/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_meta
Try mirror server https://repo.istoreos.com/repo
Installing app-meta-tailscale (0.0.4-1) to root...
Downloading https://repo.istoreos.com/repo/all/meta/app-meta-tailscale_0.0.4-1_all.ipk
Installing tailscaled (1.32.3-1) to root...
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/packages/tailscaled_1.32.3-1_x86_64.ipk
Installing tailscale (1.32.3-1) to root...
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/packages/tailscale_1.32.3-1_x86_64.ipk
Installing luci-app-tailscaler (0.0.4) to root...
Downloading https://repo.istoreos.com/repo/all/nas_luci/luci-app-tailscaler_0.0.4_all.ipk
Configuring tailscaled.
logtail started
Program starting: v1.32.3-1 (OpenWrt), Go 1.19.13: []string{"/usr/sbin/tailscaled", "--cleanup"}
LogID: 0ad61af3c4c9ea875abf7b8905c3e33b38c323a6141e23785025210bcd78d484
logpolicy: using system state directory "/var/lib/tailscale"
logpolicy.ConfigFromFile /var/lib/tailscale/tailscaled.log.conf: open /var/lib/tailscale/tailscaled.log.conf: no such file or directory
logpolicy.Config.Validate for /var/lib/tailscale/tailscaled.log.conf: config is nil
dns: [rc=unknown ret=direct]
dns: using *dns.directManager
flushing log.
logger closing down
logtail: upload: log upload of 280 bytes compressed failed: Post "https://log.tailscale.io/c/tailnode.log.tailscale.io/c06a3ce181d560e397b2c073be5e5157fbba3b60ca563b95f75809f7c67f5f52": context canceled
Configuring tailscale.
Configuring luci-app-tailscaler.
logtail started
Program starting: v1.32.3-1 (OpenWrt), Go 1.19.13: []string{"/usr/sbin/tailscaled", "--cleanup"}
LogID: 0ad61af3c4c9ea875abf7b8905c3e33b38c323a6141e23785025210bcd78d484
logpolicy: using system state directory "/var/lib/tailscale"
dns: [rc=unknown ret=direct]
dns: using *dns.directManager
flushing log.
logger closing down
Configuring app-meta-tailscale.
Try mirror server https://repo.istoreos.com/repo
Package tailscaled (1.32.3-1) installed in root is up to date.
Package tailscale (1.32.3-1) installed in root is up to date.
Package luci-app-tailscaler (0.0.4) installed in root is up to date.
```
iStore里的某些应用不是由iStore的软件源服务器提供，某些软件包iStore软件源只提供了app-meta软件包和luci软件包，其余的都是从OpenWrt官方的分支版本软件源里提供的，`tailscale`就是如此，你可以看到上面的完整安装日志。我把重点部分下面再发一遍
```
Installing app-meta-tailscale (0.0.4-1) to root...
Downloading https://repo.istoreos.com/repo/all/meta/app-meta-tailscale_0.0.4-1_all.ipk
Installing tailscaled (1.32.3-1) to root...
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/packages/tailscaled_1.32.3-1_x86_64.ipk
Installing tailscale (1.32.3-1) to root...
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.6/packages/x86_64/packages/tailscale_1.32.3-1_x86_64.ipk
Installing luci-app-tailscaler (0.0.4) to root...
Downloading https://repo.istoreos.com/repo/all/nas_luci/luci-app-tailscaler_0.0.4_all.ipk
Configuring tailscaled.
```
可以看到，从iStore里安装`tailscale`，`luci-app-tailscaler` 是由iStore软件源提供的，因为官方的`tailscale`是不提供luci的，luci-app就是网页后台，如果没有luci-app，这个软件你只能通过终端去使用。软件本体的`tailscale`和`tailscaled`就是从官方22.03.6软件源直接安装，版本为`1.32.3-1`，不由iStore软件源提供。官方22.03.6是23年12月发布的，所以版本也只能停留在那个时间点的。而且OpenWrt官方不会时时都使用最新的源码进行编译，可能在23年12月那个时间点来看，1.32.3-1都不算新。

99477204, 2025-07-09:
> ```
> Fetch feed list for x86_64
> Update feeds index
> Downloading https://istore.linkease.com/repo/all/nas_luci/Packages.gz
> Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_nas_luci [...]