istoreos/istoreos #1734: ISTOREOS 不支持NTFS格式硬盘 NFS分享吗？

lxtdd, 2024-08-01:
我入手一个X86小主机（戴尔wyse3040 ），刷了istoreos（22.03.6-2024071911-x86-64-squashfs-combined-efi.img），出现三个异常：一是小主机smb共享设置正常后，过几个小时电脑上突然无法找到这个网络共享（而且是家里所有电脑都找不到），istore小主机重启也没用，把网络共享删去，重新添加启用又好了。第二，安装nfs共享插件时（也可能是在在升级插件？），NFS管理界面突然没了，重启设备也找不回，只能重置设备到出厂状态。第三个更要命，我是利用ISTOREOS来架设NFS服务的，因为我的碟机不支持SMB1.0以上共享，严重依赖NFS服务。我的移动硬盘（NTFS格式）接入istore小主机，挂载文件系统选择ntfs-3g或者antfs，都可以开机自动挂载 ，这点比OPENWRT友好多了。把移动硬盘挂载到 /mnt/usb2-1 ，我将 /mnt/usb2-1设为NFS共享目录，设置成功后，碟机上可以搜到这个目录，却打不开长时间转圈，在电脑上用CMD命令连接，提示网络错误53 也就是找不到网络途径。于是我手动把移动硬盘分别挂载到mnt/mydisk、mnt/sda1或者mnt/sda1/aa 等目录，故障依旧。我在ISTORE系统还安装了CD2，挂载了云盘文件夹cloudnas，NFS分享这个云盘文件夹，不管哪个设备访问都一切正常。用SMB网络共享，云盘文件夹和移动硬盘也都一切正常。之前用章鱼星球刷了定制版 openwrt（带istore界面）挂载移动硬盘，只要移动硬盘成功挂载本地了，那么NFS共享出去也必然成功了。请问是ISTOREOS 不支持NTFS格式硬盘 NFS分享了吗？ 后期会有改进吗？

jjm2473, 2024-08-02:
不了解NFS，这东西版本太多还兼容性差。

不过你也不需要指定NTFS的驱动，使用默认就行了，默认是ntfs3，注意不是ntfs-3g。ntfs3是在linux下性能和功能都最完善的NTFS驱动。

closed 2024-08-02 by jjm2473

lxtdd, 2024-08-02:
好的 我试试 谢谢啦！我的情况是 碟机是主力播放器，这玩意只支持SMB1.0，所以重度依赖 NFS共享

jjm2473, 2024-08-02:
istoreos也可以提供SMB1.0服务啊，就是在samba配置里面勾选“允许旧协议与身份验证(不安全).”，默认位置 http://192.168.100.1/cgi-bin/luci/admin/services/samba4。

lxtdd, 2024-08-03:
> 不过你也不需要指定NTFS的驱动，使用默认就行了，默认是ntfs3，注意不是ntfs-3g。ntfs3是在linux下性能和功能都最完善的NTFS驱动。

好的太感谢了  经测试确实SMB4 版本可以兼容1.0，不折腾NFS了，但是我1楼陈述的第一个问题依然存在，也就是SMB4共享 仍然会闲置半小时左右就自动关闭，重启istoreos 没用，只能删除共享，重建服务，请问如何去设置啊？是不是因为我的移动硬盘自动休眠导致的？

jjm2473, 2024-08-03:
samba自动关闭？你确定是服务端关了吗？看看系统日志。还有看看客户端有没有日志。硬盘休眠应该不至于断开samba，要排查也简单，你创建个共享文件夹到/root/share之类的，就不会使用硬盘了。

lxtdd, 2024-08-04:
> samba自动关闭？你确定是服务端关了吗？看看系统日志。还有看看客户端有没有日志。硬盘休眠应该不至于断开samba，要排查也简单，你创建个共享文件夹到/root/share之类的，就不会使用硬盘了。

[image]
[image]
出现症状时，我不确定 ISTORE的samba自动关闭，看日志应该是打开的，
Sun Aug  4 09:24:49 2024 daemon.err nmbd[18936]:   nmbd version 4.18.8 started.
Sun Aug  4 09:24:49 2024 daemon.err nmbd[18936]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Sun Aug  4 09:24:49 2024 daemon.err smbd[18937]: [2024/08/04 09:24:49.637750,  0] ../../source3/smbd/server.c:1746(main)
Sun Aug  4 09:24:49 2024 daemon.err smbd[18937]:   smbd version 4.18.8 started.
Sun Aug  4 09:24:49 2024 daemon.err smbd[18937]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Sun Aug  4 09:25:55 2024 cron.err crond[19474]: crond (busybox 1.35.0) started, log level 5
Sun Aug  4 09:25:55 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sun Aug  4 09:25:56 2024 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Sun Aug  4 09:25:56 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Sun Aug  4 09:25:56 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset no-nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sun Aug  4 09:25:56 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Sun Aug  4 09:25:56 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sun Aug  4 09:25:56 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sun Aug  4 09:25:56 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost 

总之几次出现SMB网络共享问题都和硬盘有关，上次是怀疑硬盘盒休眠了 不会自动醒，我索性安装了istore商店的 硬盘休眠app,应该有 唤醒硬盘功能，果然长时间SMB正常运行。就刚才 我挂载的移动硬盘忘了通电，PC上打开网络共享 就是打不开，如截图所示，我于是把共享中硬盘删掉，只剩下云盘共享，果然正常了。 是不是说，SMB要是找不到它共享的资源，就会拒绝共享服务？我是小白  纯属瞎猜。

lxtdd, 2024-08-09:
我发现了，只有勾选“允许旧协议与身份验证(不安全).”  才会出现 硬盘没挂载，网络共享打不开的情况，不勾选一切正常  不纠结了。另外，最新版的istore已经解决了 ntfs格式硬盘 NFS共享问题  大大的赞一个！！