istoreos/istoreos #1740: 为什么电脑一直有一个6.6.6.6的IP

dcluffy, 2024-08-04:
## 详细叙述

### (1) 具体问题

A：istoreos刚装没两天，下面的一台电脑一直多一个6.6.6.6的IP，搞不懂这是哪儿来的，以前的istoreos固件也有这个问题，也是这台电脑，一直想问一下这是什么情况，有大佬可以解惑吗

### (2) 路由器型号和固件版本

A：Intel Geminilake
iStoreOS 22.03.6 2024062810

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-09-04:
根据mac地址到对应设备上看看它是不是有这个ip地址呗

daoyelihuowang, 2024-10-22:
我也遇到了这个问题都是一个mac地址但会多出来很多奇怪的ip，跟设置的内网网段根本不上一个网段而且根据mac地址查找ip地址没有那些ip

dcluffy, 2024-10-23:
> 我也遇到了这个问题都是一个mac地址但会多出来很多奇怪的ip，跟设置的内网网段根本不上一个网段而且根据mac地址查找ip地址没有那些ip

你这个好像跟我的情况不太一样，我这个后来查出来是电脑内存导致的，更换内存后那台电脑再也没有其他ip出现了

daoyelihuowang, 2024-10-24:
> > 我也遇到了这个问题都是一个mac地址但会多出来很多奇怪的ip，跟设置的内网网段根本不上一个网段而且根据mac地址查找ip地址没有那些ip
> 
> 你这个好像跟我的情况不太一样，我这个后来查出来是电脑内存导致的，更换内存后那台电脑再也没有其他ip出现了

我是小白没搞懂，电脑内存应该跟网络ip没什么关联吧 ，请解惑一下！

dcluffy, 2024-10-26:
> > > 我也遇到了这个问题都是一个mac地址但会多出来很多奇怪的ip，跟设置的内网网段根本不上一个网段而且根据mac地址查找ip地址没有那些ip
> > 
> > 
> > 你这个好像跟我的情况不太一样，我这个后来查出来是电脑内存导致的，更换内存后那台电脑再也没有其他ip出现了
> 
> 我是小白没搞懂，电脑内存应该跟网络ip没什么关联吧 ，请解惑一下！

我比你也强不到哪儿去，我是因为电脑玩游戏自动关机排查故障过程中发现有一根内存出问题了，去掉这根内存就好了，这个时候恰巧发现istoreos没有分配6.6.6.6这个IP了，你要我说出个所以然来，我也不懂

closed 2024-10-26 by dcluffy

jjm2473, 2024-10-28:
在电脑上抓包看看有没有发出源地址是6.6.6.6的数据包吧。另外我怀疑是不是跟某些翻墙插件的fakeip啥的有关，你关掉翻墙插件后，把软路由和电脑都重启下看看。