istoreos/istoreos #1741: 求助：istoreos启用ipv6成功，公网能正常访问，但局域网设备无法访问公网ipv6

jagerzhang, 2024-08-04:
移动宽带接入，已启用ipv6，从公网可以访问到istoreos上的服务。
通过设置，也成功让macbook能够分配到了ipv6：
[image]

从macbook上能通ipv4，但无法ping6通公网ipv6地址：
[image]

看起来ipv6的解析是正常的，疑似路由问题，于是我试了下，从macbook ping6 网关也提示没有路由
[image]

接触istoreos/openwrt时间不长，很多设置尤其是网络这块都没能弄明白，求大神帮忙看看，感激不尽。

138161, 2024-08-05:
老问题，无PD的ipv6，已解决，搜索下我的帖子，只有2种方法：NAT和中继
上海电信IPV6_NAT方式v2
上海电信IPV6光猫拨号ipv6中继方式
链接: https://pan.baidu.com/s/1lzTFTQF3GrFiOXgaBlrvSA
提取码: wewk

jagerzhang, 2025-01-04:
现在确认成功获取了 ipv6-pd:
[image]

软路由本身也可以ping6 通  ipv6 ip 或网址:
[image]

但是局域网，比如 mac os  能解析，但是无法 ping 通 ipv6：
[image]

jagerzhang, 2025-01-06:
结贴，其实配置都没问题，主要问题出现在华为AP160对ipv6的支持有问题，太坑了.. 换个其他普通AP都可以。

closed 2025-01-06 by jagerzhang