istoreos/istoreos #1744: 重启后，登录不了luci的界面

stephenlzc, 2024-08-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：重启后，无法登录luci界面，

### (2) 路由器型号和固件版本

A：路由器为x86，J1900双千兆网卡，固件版本为：istoreos-22.03.6-2024071911-x86-64-squashfs-combined.img

### (3) 详细日志和/或截图

A：当我重新启动路由器后，无法通过网页的方式登录luci界面，但ssh登录是正常的。

[image]

提示如下：
Status: 500 Internal Server Error
> /usr/lib/lua/luci/template.lua:97: Failed to execute template 'quickstart/home'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'quickstart/main'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/argon/header'. A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: attempt to index local 'boardinfo' (a nil value) stack traceback: [string "/usr/lib/lua/luci/view/themes/argon/header...."]:36: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function 'render' /usr/lib/lua/luci/dispatcher.lua:699: in function 'include' [string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function 'render' /usr/lib/lua/luci/dispatcher.lua:699: in function 'include' [string "/usr/lib/lua/luci/view/quickstart/main.htm"]:1: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function (tail call): ? [string "/usr/lib/lua/luci/view/quickstart/home.htm"]:10: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function (tail call): ? /usr/lib/lua/luci/dispatcher.lua:1052: in function 'dispatch' /usr/lib/lua/luci/dispatcher.lua:482: in function

[image]

Vash90, 2024-09-14:
跟你遇到 非常类似的问题。刚装好的 istoreos 重启都正常，用一段时间装过一些东西后，重启就没法进入首页。
只能断电后 启动会正常。

alanzhu2019, 2024-11-02:
遇到一样的问题，只要重启就无法访问，必须断电才行，，有哪位大佬能解这个问题!!!

stephenlzc, 2024-11-03:
@juhosg @ffainelli @jjm2473

jjm2473, 2024-11-07:
x86的话，试试bios禁用快速启动之类的

LeonsCarry, 2024-12-10:
同样的问题，不知道怎么解决

LeonsCarry, 2024-12-12:
我遇到相同的问题，已解决。楼主是不是接了机械硬盘，我把我的机械硬盘拿掉就一切正常了，原因是机械硬盘盒有故障，导致系统加载不了硬盘信息，无法进入系统，楼主可以检查一下机械硬盘或者硬盘盒，又或者检查一下硬盘是否有故障，应该就可以解决了

LeonsCarry, 2024-12-12:
> 遇到一样的问题，只要重启就无法访问，必须断电才行，，有哪位大佬能解这个问题!!!

看我的回复