istoreos/istoreos #1758: 同样设置 istore无法连接YT,openwrt畅通

lxtdd, 2024-08-09:
passwall ，ISTOREOS 连接大部分网站（github等）非常正常，却无法连接 油管  
“未连接：有潜在的安全问题  Firefox 检测到潜在的安全威胁，并因 www.youtube.com 要求安全连接而没有继续。
您可以做什么？www.youtube.com 启用了被称为 HTTP 严格传输安全（HSTS）的安全策略，Firefox 只能与其建立安全连接。您无法为此网站添加例外，以访问此网站。这个问题大多与网站有关，无法通过您的操作解决。您可以向此网站的管理者反馈此问题。”

奇怪的是，换openwrt ，同样的机场票  同样passwall  相同设置  油管 畅通，请指教。

lxtdd, 2024-08-09:
> passwall ，ISTOREOS 连接大部分网站（github等）非常正常，却无法连接 油管 “未连接：有潜在的安全问题 Firefox 检测到潜在的安全威胁，并因 [www.youtube.com](http://www.youtube.com) 要求安全连接而没有继续。 您可以做什么？www.youtube.com 启用了被称为 HTTP [...]

换其他浏览器 亦如此

lxtdd, 2024-08-09:
参考openwrt防火墙及网络设置后 已经解决  请管理员关闭主题

jjm2473, 2024-09-04:
是否passwall配置了使用firewall4？istoreos目前使用的是firewall3

closed 2024-09-04 by jjm2473