istoreos/istoreos #1762: iStoreOS docker pull 和 push 私有仓库不成功，daemon.json配置不生效

dengwentong, 2024-08-14:
## 1.iStoreOS docker pull 和 push 私有仓库不成功
**istoreOS版本：** 22.03

## 2. 详细叙述
已经配置了 /etc/docker/daemon.json 并且也已经重新启动docker和路由器了，依然不生效。
我在公司服务器 CentOS下，Ubuntu下均能成功。

### (1) 具体问题
✘ ⚡ root@iStoreOS  /usr/lib  docker pull www.123.org:8050/dwt-harbor/redis-photon:v2.10.3
Error response from daemon: Get "https://www.123.org:8050/v2/": http: server gave HTTP response to HTTPS client

[image]
[image]

dengwentong, 2024-08-21:
已经找到方法了。

jjm2473, 2024-09-04:
istoreos的docker默认不是使用这个daemon.json，而是使用生成的。

albicocccca, 2024-11-07:
> istoreos的docker默认不是使用这个daemon.json，而是使用生成的。

我也遇到这个情况了，大哥能说的具体些openwrt上怎么操作吗，换源不生效

albicocccca, 2024-11-07:
> 已经找到方法了。

能教教我吗

jjm2473, 2024-11-07:
@albicocccca 删掉/etc/config/dockerd，然后自己写好 /etc/docker/daemon.json ，然后执行`/etc/init.d/dockerd restart`就行了。不过不推荐自己写daemon.json，毕竟istoreos本来就提供了后台页面管理docker配置。还有如果自己写daemon.json，注意禁用掉docker的iptables功能，以免出现安全问题。

londbell, 2025-06-02:
> > [@albicocccca](https://github.com/albicocccca) 删掉/etc/config/dockerd，然后自己写好 /etc/docker/daemon.json ，然后执行`/etc/init.d/dockerd restart`就行了。不过不推荐自己写daemon.json，毕竟istoreos本来就提供了后台页面管理docker配置。还有如果自己写daemon.json，注意禁用掉docker的iptables功能，以免出现安全问题。
> 
> 主路由istoreos安装了kspeeder，打开[http://192.168.2.1:5003可以看到有效](http://192.168.2.1:5003%E5%8F%AF%E4%BB%A5%E7%9C%8B%E5%88%B0%E6%9C%89%E6%95%88)
> 
> 在旁路由的istoreos上设定了镜像服务器 Registry Mirrors: https://192.168.2.1:5443/
> 
> 结果pull依然报错： docker pull adolfintel/speedtest:latest Error response from daemon: Get "https://registry-1.docker.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)

在主路由将registry.linkease.net解析到192.168.2.1；然后将镜像地址改成https://registry.linkease.net:5443/，问题解决，为啥不可以使用ip形式？