istoreos/istoreos #1769: 请问如何限制ipv6获取到的地址只有一个

wqinf, 2024-08-17:
请问如何限制ipv6获取到的地址只有一个？目前通过默认的ipv6设置，每台设备获取到了多个ipv6地址。
之前用TP路由器的时候，ipv6地址获取方式设置为SLAAC的时候，每台设备就能获取到1个ipv6地址。
请问在istoreos中若想实现类似的效果，应该怎么实现？不敢瞎搞，怕搞坏了，请大佬指导一下，谢谢！

138161, 2024-08-18:
SLAAC方式，都是由上级网关决定的，比如光猫决定，都是多个ipv6地址，一个是根据网卡地址算出来的，其他的是光猫分配的临时ipv6地址，光猫下面的客户端无法改变，决定权在光猫，不在下面的客户端。
NAT6方式，由光猫下面的路由器决定ipv6地址的分配，那就不是240e开头，是fd00啥的开头（自己定义），相当于ipv6下的局域网，这个时候ipv6的临时地址才能关闭。
当然，以上都是运营商未下发PD前缀的情况，如果PD前缀下发，并且小于64位，意味着，光猫下的路由器能代替光猫管理ipv6地址，这就另当别论了。

wqinf, 2024-08-18:
> SLAAC方式，都是由上级网关决定的，比如光猫决定，都是多个ipv6地址，一个是根据网卡地址算出来的，其他的是光猫分配的临时ipv6地址，光猫下面的客户端无法改变，决定权在光猫，不在下面的客户端。 NAT6方式，由光猫下面的路由器决定ipv6地址的分配，那就不是240e开头，是fd00啥的开头（自己定义），相当于ipv6下的局域网，这个时候ipv6的临时地址才能关闭。 当然，以上都是运营商未下发PD前缀的情况，如果PD前缀下发，并且小于64位，意味着，光猫下的路由器能代替光猫管理ipv6地址，这就另当别论了。

目前我这里是将软路由作为主路由用的，有办法实现限制ipv6获取到的地址只有一个吗？谢谢