istoreos/istoreos #1773: IPV6突然造成国内网站访问不正常，有的能访问有的不能，有谁有经验的帮忙解决么

aksehf123, 2024-08-19:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
没有类似的issue

详细叙述：
光猫桥接，x86的istore路由拨号。电信宽带。

之前按照小宝的ipv6开启教程能正常使用，且解析了动态域名也是正常（大概用了3个月左右都一切正常）。
但是前几天不小心把wan口的网线插错了再插回去就开始不正常（症状：显示软件源配置错误，国外能正常访问，国内baidu ipv4和ipv6都无法ping通，其他有些常用国内网站能访问但是速度很慢）。

将路由重置，只进行拨号联网，其他什么都没设置的情况下还是网络不正常且显示软件源配置错误，只能用系统便捷工具把ipv6彻底关闭，所有网络才正常访问；完全按照小宝的教程重新配置ipv6还是问题依旧。

★最关键的是我重置系统什么设置都不做只进行拨号上网的情况下，网络访问都不正常，只能通过完全关闭ipv6，但是又有使用ipv6的环境需要。

目前只能暂时彻底关闭ipv6使用，有没有大佬遇到过相似情况的有什么解决方案么。

路由器型号和固件版本：
x86，
固件iStoreOS 22.03.7 2024080210

目前工作环境没在家无法提供截图和日志，只是以上信息看有没有兄台有解决方法

138161, 2024-08-31:
ipv6没配置好，就会有上面的情况，如何配置ipv6，确实很难简单说清楚，电信的网络，目前大概率是给个ipv6地址，无PD前缀的，所以，配置只有2种方式，可参考下面方法
https://github.com/istoreos/istoreos/issues/1480

138161, 2024-08-31:
链接: https://pan.baidu.com/s/1fegI9ZTLz62tB2v1USyLkA 提取码: fqi6 复制这段内容后打开百度网盘手机App，操作更方便哦
链接: https://pan.baidu.com/s/14wWW0Gz_ckxo_c18n1L6Cw 提取码: f3q8 复制这段内容后打开百度网盘手机App，操作更方便哦

jjm2473, 2024-09-04:
光猫和软路由都重启下试试

452138222, 2024-11-27:
我也遇到，难搞