istoreos/istoreos #1779: 麻烦能帮我看下为什么自动重启了吗？谢谢

wqinf, 2024-08-22:
[日志.txt](https://github.com/user-attachments/files/16705670/default.txt)

wqinf, 2024-09-05:
经过排查，发现问题集中在网速跑满的时候，大概率出现重启问题。
排查思路：
1、更换更高功率的电源适配器，因为之前换了风扇，还单独在USB接口插了一个散热底座，不过换了更高功率的电源适配器问题依旧；
2、系统重装，问题依旧；
3、保持系统默认设置，再依次排查插件，最后注意到从商店安装的微信推送版本比较旧，然后从其Github发布页面下载手动安装了最新版本的就再没出现问题了，新旧系统之间这有这个差异，因此可以基本判定是这个插件太旧导致的。

如果也遇到了这个问题的朋友们可以参考一下，如果解决了问题记得跟帖分享一下。

也希望istore商店能够及时更新相关的插件，部分插件已经很久没更新了。

closed 2024-09-09 by wqinf

shanhai2333, 2025-03-27:
> 经过排查，发现问题集中在网速跑满的时候，大概率出现重启问题。 排查思路： 1、更换更高功率的电源适配器，因为之前换了一个风力更强进的风扇，还单独在USB接口插了一个散热底座，怀疑是功率不足，不过换了更高功率的电源适配器问题依旧； 2、系统重装，在网速跑满的时候依然会重启，问题依旧； 3、保持系统默认设置，再依次排查插件，直到排查到最后一个插件，也就是从商店安装的微信推送这个插件的版本比较旧，然后从其Github发布页面下载手动安装了最新版本，安装上之后在网速跑满的时候也没再出现问题了，新旧系统之间这有这个差异，因此可以基本判定是这个插件太旧导致的。
> 
> 如果也遇到了这个问题的朋友们可以参考一下，如果解决了问题记得跟帖分享一下。
> 
> 也希望istore商店能够及时更新相关的插件，部分插件已经很久没更新了，比如微信推送、Lucky等。

感谢您，赛博神医，困扰我三个月的问题解决了，我猜也是哪个插件问题，但是不知道怎么排查，重启也看不到日志，结果是推送的锅

wqinf, 2025-03-28:
> 感谢您，赛博神医，困扰我三个月的问题解决了，我猜也是哪个插件问题，但是不知道怎么排查，重启也看不到日志，结果是推送的锅

当时也是考虑可能会有人遇到相同的问题，所以把排查过程记录了下来。