istoreos/istoreos #1782: 内网中NAS上ddns-go解析Nginx Proxy Manager代理的服务不能访问，求解决，谢谢了。

kiki-theone, 2024-08-25:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 我在内网有一个UnRaid的NAS，其中运行了Nginx Proxy Manager的Docker，并通过UnRaid上的ddns-go解析域名到Nginx Proxy Manager。在硬路由和openwrt.ai编译的固件下，Nginx Proxy Manager可以代理的服务可以正常使用；但是换用istoreos后，无论外网还是内网都不能通过域名访问。

尝试过在DHCP/DNS中关闭过滤器中的“重绑定保护”和安装“uHTTP”并关闭“忽略公共接口上的私有 IP”，均没有效果。
防火墙中入站、出站、转发都改为了接受，区域转发也都改为接受了。

### (2) 路由器型号和固件版本

A：软路由为X86 畅网8505和小马V1都尝试过。
固件 iStoreOS 22.03.7 2024080210 / LuCI istoreos-22.03 branch git-24.207.28853-3b34de8

kiki-theone, 2024-10-22:
这么久了也没有人提供解决办法吗？

mclarenlm, 2024-11-06:
同样的问题，同求解决办法