istoreos/istoreos #1793: openclash+群晖ahg无法配置自定义上游服务器

michu0126, 2024-09-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 我的AGH用的是群晖docker版，我的网络开启了ipv6
我的openwrt作为旁路由，ikuai作为主路由，安装在单独的主机的虚拟机里
我在op里面配置openclash的上游服务器为群晖agh，在name，fallback，default均添加udp类型，短时间内可以响应并有dns查询记录，无过滤记录，长时间open clash节点直接断连，此种情况在本地dns劫持选择三种模式下均会出现
于是我启用了op里的agh，添加op的agh作为openclash的上游服务器，此时agh的dns查询和过滤均有数据统计，运行正常，但我的网络无法访问google和YouTube

### (1) 固件可能存在相关bug

A：

### (2) 路由器型号和固件版本

A：22.03.7

### (3) 详细日志和/或截图

A：
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001job.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001jp.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001kd.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001pp.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001tech.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001town.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001tudou.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 001u.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 00257.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53 for domain 002lzj.com
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: more servers are defined but not logged
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Sep  3 01:46:12 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: private/reserved address 192.168.50.4 is not suitable for external IP
Tue Sep  3 01:48:10 2024 daemon.notice miniupnpd[15686]: force_forwarding enable, private/reserved address 192.168.50.4 used as external IP

jjm2473, 2024-09-04:
openclash本身就很多bug，出什么事都不意外。

zhx47, 2024-11-24:
> openclash本身就很多bug，出什么事都不意外。

有什么推荐的吗，我之前也在用openclash，最近也会出现只要启动openclash启动，大概一天就无法访问任何域名。但是没有科学的话docker有很麻烦了

zhx47, 2024-11-25:
> istore

感谢，因为我会把软路由当作测试机，这几天最让我头大的就是写好的东西打docker镜像的时候基础镜像的问题了，还有开发的一些乱七八糟的依赖