istoreos/istoreos #1800: 最新版istoreos x86-64无法识别nvme硬盘

PingCd, 2024-09-04:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法识别nvme硬盘

### (2) 路由器型号和固件版本

A：x86-64-efi 22.03.7

### (3) 详细日志和/或截图

A：
[image]

PingCd, 2024-09-07:
有官方人员么？

cgfyl, 2024-09-16:
我也遇到类似的问题，最后折腾了三四个小时，用1月份版本就好了。
[image]
[image]

PingCd, 2024-09-16:
多谢！

jjm2473, 2024-09-24:
机型是什么，还有硬盘型号是什么

PingCd, 2024-09-24:
cpu:j3160
主板：海尔
内存：ddr3l 4g * 2
pcie转nvme：三星
[image]
[image]

jjm2473, 2024-09-26:
开机日志都跑到40多秒，肯定启动完了。按下回车，然后执行`lspci -nnk`看看

ITGER-coder, 2025-03-07:
一样的问题，最后解决了吗？

PingCd, 2025-03-07:
最后换了一根SSD解决，估计认硬盘