istoreos/istoreos #1803: 统一文件共享丢失

ymsq, 2024-09-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在统一文件共享内设置了samba，添加了用户，搬家后发现共享无法访问，检查发现该入口丢失。

### (2) 路由器型号和固件版本

A：[Powered by LuCI openwrt-22.03 branch (git-24.208.65546-18f3914)](https://github.com/openwrt/luci) / [ArgonTheme v2.2.10.10](https://github.com/jerrykuku/luci-theme-argon) / iStoreOS 22.03.7 2024080210

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-09-24:
这是怎么发生的，菜单丢失前做了啥操作

ymsq, 2024-10-06:
> 这是怎么发生的，菜单丢失前做了啥操作

搬家时移除了路由器电源，再次上电开机后丢失