istoreos/istoreos #1808: 插件下载不了了，404

woeoio, 2024-09-11:
反馈bug/问题模板，提建议请删除

https://[istore.linkease.com](https://istore.linkease.com/)/

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

girlmn, 2024-09-14:
1. 下载最新固件时 下载网页发生错误
2. r5s 
3. 
[image]
[image]

girlmn, 2024-09-14:
请尽快恢复

jjm2473, 2024-09-24:
不是还有其他备用地址吗？或者到这里下载 https://sourceforge.net/projects/istoreos/files/iStoreOS/

jjm2473, 2024-09-24:
而且这不是插件，是固件。而且不是404，是连接错误。

closed 2024-09-24 by jjm2473

woeoio, 2024-09-24:
后来好了