istoreos/istoreos #1817: 关于115网盘能挂载的alist需要更新最新版，谢谢！

hnzhaihui, 2024-09-17:
115服务器崩溃后恢复，alist不能登录挂载的115网盘了，本人的安装的iStoreOS 22.03.6 2024032917 ，商店里的alist还是3.33.0，最新alist是3.37.3，已经解决了115网盘不能登录的问题，希望能尽快更新。
或者有大佬教一下安装的商店里的软件，是否能自己下载alist最新版本，更新3.37.3不？有教程最好!
谢谢！

hnzhaihui, 2024-09-17:
因为有大佬说不能自己pull更新系统商店里的软件，有可能会导致istoreos系统崩溃：                                                         
 不要升级。我们已经在软件中心更新了 alist。
建议用沙箱模式，崩溃可以马上回退到上次好的版本。
系统软件包不要升级，他们有依赖关系，升级之后，可能因为软件依赖问题，系统崩溃了！                   

所以希望能尽快更新alist到3.37.3，感谢ing！！！

hnzhaihui, 2024-09-17:
另外CD2释放出的修改json文件的办法，我估计用来更新alist的json文件也能成功，就是不知道alist的json文件在istroeos里的那个目录里，有大佬试过么！

hnzhaihui, 2024-09-17:
或者哪位大神是否可以告知 在istroeos中商店里安装的软件的安装目录，我自己想办法去更改alist登录115的UA，就可以解决这个问题，谢谢！

xiaomeng9597, 2024-09-17:
直接用二进制文件运行不就好了，我从来不用alist的ipk安装包，并且ipk更新太麻烦，另外这玩意更新太频繁了，二进制文件方式，只需要更新这个二进制文件重新运行一下，或者重启系统就行，需要在/etc/rc.local文件里面加入后台启动命令

下面是命令加入到/etc/rc.local文件里面就行，/mnt/mmcblk0p18/alist 是目录，alist目录里面有alist文件，第三条是后台启动命令并且在此文件夹里面实时生成运行log.txt日志。

cd /mnt/mmcblk0p18/alist
chmod -R 777 alist
./alist server > log.txt 2>&1 &

[image]
[image]

KoreAkira, 2024-09-17:
我已经手动升级了3.37.3暂时无问题,升级前注意几个问题，提前备份，确认自己系统版本，CPU核心号，一定不能搞错，https://github.com/sbwml/luci-app-alist/releases/tag/v3.37.3

KoreAkira, 2024-09-18:
不过好像这个版本也有问题，不能通过Alist复制其他网络盘到115盘

hnzhaihui, 2024-09-18:
> 不过好像这个版本也有问题，不能通过Alist复制其他网络盘到115盘

能访问就不错了，我也暂时没跨盘复制的需求，谢谢，我等下试试！

KoreAkira, 2024-09-18:
后续还是等吧，应该会更新这个bug。

hnzhaihui, 2024-09-18:
> 我已经手动升级了3.37.3暂时无问题,升级前注意几个问题，提前备份，确认自己系统版本，CPU核心号，一定不能搞错，https://github.com/sbwml/luci-app-alist/releases/tag/v3.37.3

我是Phicomm N1刷的iostoreos 版本22.03，架构ARMv8 Processor rev 4，查看现有alist是v3.33.0-aarch64，那我要选择[openwrt-22.03-aarch64_cortex-a53.tar.gz]还是[openwrt-22.03-aarch64_cortex-a72.tar.gz]呢？谢谢

hnzhaihui, 2024-09-18:
> 后续还是等吧，应该会更新这个bug。

嗯，先能挂载访问，以后再等更新，我查了一下ARMv8好像都是a53架构，等下升级试一下，再次感谢大佬！

hnzhaihui, 2024-09-18:
> 我是Phicomm N1刷的iostoreos 版本22.03，架构ARMv8 Processor rev 4，查看现有alist是v3.33.0-aarch64，那我要选择

谢谢大佬回复指导，我是Phicomm N1刷的iostoreos 版本22.03，架构ARMv8 Processor rev 4，查看现有alist是v3.33.0-aarch64，那我要选择二进制包的那个才行，谢谢！

hnzhaihui, 2024-09-18:
> 直接用二进制文件运行不就好了，我从来不用alist的ipk安装包，并且ipk更新太麻烦，另外这玩意更新太频繁了，二进制文件方式，只需要更新这个二进制文件重新运行一下，或者重启系统就行，需要在/etc/rc.local文件里面加入后台启动命令。
> 
> 二进制文件下载地址：https://github.com/alist-org/alist/releases，下载最新自己机型相对应CPU架构的压缩包并且解压到下面你自定义的目录里面。
> 
> 下面是命令加入到/etc/rc.local文件里面就行，/mnt/mmcblk0p18/alist 是目录需要自行修改，alist目录里面有alist文件，第三条命令是后台启动alist二进制文件并且在此目录里面实时生成运行log.txt日志。
> 
> cd /mnt/mmcblk0p18/alist chmod -R 777 alist ./alist server > log.txt 2>&1 &
> 
> ![Screenshot_2024-09-18-06-09-01-281_bin mt plus](https://private-user-images.githubusercontent.com/49544781/368368390-82722362-e1f7-4020-8f3d-692e3bdcf3e8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2NDM5NTgsIm5iZiI6MTcyNjY0MzY1OCwicGF0aCI6Ii80OTU0NDc4MS8zNjgzNjgzOTAtODI3MjIzNjItZTFmNy00MDIwLThmM2QtNjkyZTNiZGNmM2U4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDA3MTQxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5MjMwNjc4MzAwN2FlNmE0ZjNiZTZkNDNjM2RhMGEzOWZmYTIyMDNmOWFiMTAxNjM0NTY4YWY2ZGZhNWE5NjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OkuGD_CJ4pOGeuBvRRafb0ZwryWPaWOZDjxFZwWpnQo) ![Screenshot_2024-09-18-06-11-20-862_bin mt plus](https://private-user-images.githubusercontent.com/49544781/368359149-d9f0a294-9be1-4833-b39f-d1780aaabfef.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2NDM5NTgsIm5iZiI6MTcyNjY0MzY1OCwicGF0aCI6Ii80OTU0NDc4MS8zNjgzNTkxNDktZDlmMGEyOTQtOWJlMS00ODMzLWIzOWYtZDE3ODBhYWFiZmVmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDA3MTQxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1ODA4NjYzZjZjMzY1ZWFmMjJiYjZhN2YyNzA2ZmMxMDBjNGE1ZDE1ODIxMDM3Zjk0NWM4MmZiMjJmYjhkMTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.eWcJqoS5bGdI47oA7bPbUP4xcaXCmLZDyVRjp391fjM)

另外，我的alist是在istroeos中商店里安装的，所以安装目录没在mnt里，但是cd2安装目录在这里，alist不好找，正在搜索中。。。

hnzhaihui, 2024-09-18:
> 直接用二进制文件运行不就好了，我从来不用alist的ipk安装包，并且ipk更新太麻烦，另外这玩意更新太频繁了，二进制文件方式，只需要更新这个二进制文件重新运行一下，或者重启系统就行，需要在/etc/rc.local文件里面加入后台启动命令。 [...]

[image]
搜了一下整个iostoreos的目录，60多M大小的alist有几十个，那就不好判断那个才是正在运行的alist包了。

KoreAkira, 2024-09-19:
37.4已经修复115的Bug

hnzhaihui, 2024-09-20:
> 又更新了37.4不知道115上传问题解决了没

alist又更新了？我刚更新了37.3，确实我群晖里cloudsync搞的自动备份NAS文件到115确实一直报错，37.4如果真的解决了，那我试试。

WangDefa88, 2024-09-20:
> 我已经手动升级了3.37.3暂时无问题,升级前注意几个问题，提前备份，确认自己系统版本，CPU核心号，一定不能搞错，https://github.com/sbwml/luci-app-alist/releases/tag/v3.37.3

请问你下载的gz文件是怎么放到istore里面安装的呀？我看它只支持ipk和run文件

WangDefa88, 2024-09-20:
> alist又更新了？我刚更新了37.3，确实我群晖里cloudsync搞的自动备份NAS文件到115确实一直报错，37.4如果真的解决了，那我试试。 查了一下，可惜我的istroeos内核是22.03 现在aarch64版本的3.37.4只有22.02，不知道能升不

请问你下载的gz文件是怎么放到istore里面安装的呀？我看它只支持ipk和run文件

hnzhaihui, 2024-09-20:
> > alist又更新了？我刚更新了37.3，确实我群晖里cloudsync搞的自动备份NAS文件到115确实一直报错，37.4如果真的解决了，那我试试。 查了一下，可惜我的istroeos内核是22.03 现在aarch64版本的3.37.4只有22.02，不知道能升不
> 
> 请问你下载的gz文件是怎么放到istore里面安装的呀？我看它只支持ipk和run文件

要找到iostoreos的相应目录，我没找到，我是直接ipk升级的，不过你现在不用了，因为刚刚istroeos已经更新了alist的最新版3.37.4，去愉快的玩耍吧！

closed 2024-09-24 by jjm2473

nanshihui, 2025-02-21:
thank you for maintain the isotreos project.@jjm2473  . there has a new version 3.42.0 about alist. can you develop the script to auto update. by the way , in istore alist's page, i cloud not set the memory of alist. can u add it ? thanks anyway.