istoreos/istoreos #1818: Dnsmasq cname 无法使用

hsqbh, 2024-09-18:
## 1.关于你要提交的问题

Dnsmasq cname 设置域名目标 必须要浏览器访问一下目标地址
然后才能解析域名 大概过一分钟左右 又会失去解析
从新访问目标 又可以解析 一分钟后又会断开解析

### (2) 路由器型号和固件版本

型号 | Techvision TVI7309X
架构 | Intel(R) Celeron(R) N5105 @ 2.00GHz
目标平台 | x86/64
固件版本 | iStoreOS 22.03.7 2024080210 / LuCI istoreos-22.03 branch git-24.207.28853-3b34de8
内核版本 | 5.10.221

jjm2473, 2024-09-24:
详细说明下，配置的参数是什么，到底什么设备解析失败

hsqbh, 2024-09-24:
> 详细说明下，配置的参数是什么，到底什么设备解析失败

没什么配置参数阿  就DHCP/DNS选项 里面的CNAME 然后windows电脑和手机都试过 都是失败的

jjm2473, 2024-09-26:
使用nslookup测试，把测试过程发出来看看

fshx208, 2025-03-17:
我也遇到一样的问题

fshx208, 2025-03-17:
[image]

[image]

jjm2473, 2025-03-17:
### 分析
根据dnsmasq的手册 https://thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html
```
--cname=<cname>,[<cname>,]<target>[,<TTL>]
    Return a CNAME record which indicates that <cname> is really <target>. 
    There is a significant limitation on the target; it must be a DNS record which 
      is known to dnsmasq and NOT a DNS record which comes from an upstream server. 
    The cname must be unique, but it is permissible to have more than one cname pointing to the same target. 
    Indeed it's possible to declare multiple cnames to a target in a single line, like so: --cname=cname1,cname2,target

    If the time-to-live is given, it overrides the default, which is zero or the value of --local-ttl. The value is a positive integer and gives the time-to-live in seconds.
```

其中关键一句
> There is a significant limitation on the target; it must be a DNS record which is known to dnsmasq and NOT a DNS record which comes from an upstream server.

cname的目标需要是已知的地址，不能从上游 DNS 服务器解析。也就是需要先绑定目标域名的IP地址，然后才能cname到这个目标域名。

### 解决方案

#### 方案1
可以在“主机名映射”里面添加目标域名到IP地址的映射（可添加多个），之后cname就能生效了。但这种方式不适合IP地址经常变化的场景。

#### 方案2
另一个方案是使用额外的hosts文件，定期刷新这个hosts文件，让dnsmasq重新加载。
具体操作如下，先在终端中执行`touch /etc/migu_hosts.txt`创建一个空的hosts文件，然后后台页面到“Resolv 和 Hosts 文件”页签，在“额外的 HOSTS 文件”填入文件路径“/etc/migu_hosts.txt”，然后点击+号，然后保存应用。
下一步就是创建一个定时任务，例如每天或者每个小时刷新一次 /etc/migu_hosts.txt 的内容。
（未完待续。。。）

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/2043/1

jjm2473, 2025-10-24:
@wxhbts 你举例的两个参数不是一样吗？还是说你删掉了`--`？