istoreos/istoreos #1819: R2S使用USB wifi（COMFAST CF-812AC)不定时失去连接

zhukai168, 2024-09-19:
**(1) 具体问题**

   刷istoreOS后，各功能使用流畅，系统运行流畅。
   作为AP使用的wifi，启动不久就失去连接！！！
   USB WIFI型号：COMFAST CF-812AC

**(2) 路由器型号和固件版本**
    a、路由器型号：
           R2S
    b、固件版本
           istoreos-22.03.7-2024080210-r2s-squashfs.img.gz	111.9 MiB	2024-08-02 10:56

**(3) 详细日志和/或截图**
  ——————系统日志——————
--
Wed Sep 18 21:51:45 2024 daemon.notice   hostapd: wlan0: AP-STA-DISCONNECTED 8e:c2:dd:49:9b:5e
Wed Sep 18 21:51:45 2024 kern.warn   kernel: [ 1034.041028] rtw_8822bu 1-1:1.0: timed out to flush queue 3
Wed Sep 18 21:51:45 2024 daemon.info   hostapd: wlan0: STA 8e:c2:dd:49:9b:5e IEEE 802.11: disassociated
Wed Sep 18 21:51:45 2024 kern.warn   kernel: [ 1034.401040] rtw_8822bu 1-1:1.0: timed out to flush queue 3
Wed Sep 18 21:51:46 2024 daemon.info   hostapd: wlan0: STA 8e:c2:dd:49:9b:5e IEEE 802.11: deauthenticated due to   inactivity (timer DEAUTH/REMOVE)
Wed Sep 18 22:01:03 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: authenticated
Wed Sep 18 22:01:03 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: associated (aid 1)
Wed Sep 18 22:01:04 2024 kern.warn   kernel: [ 1593.089873] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:05 2024 kern.warn   kernel: [ 1594.089922] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:06 2024 kern.warn   kernel: [ 1595.089816] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:07 2024 kern.warn   kernel: [ 1595.829826] rtw_8822bu 1-1:1.0: timed out to flush queue 2
Wed Sep 18 22:01:07 2024 kern.warn   kernel: [ 1596.029924] rtw_8822bu 1-1:1.0: timed out to flush queue 3
Wed Sep 18 22:01:11 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: deauthenticated due to   local deauth request
Wed Sep 18 22:01:16 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: authenticated
Wed Sep 18 22:01:16 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: associated (aid 1)
Wed Sep 18 22:01:16 2024 kern.warn   kernel: [ 1605.369492] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:17 2024 kern.warn   kernel: [ 1606.369467] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:18 2024 kern.warn   kernel: [ 1607.369533] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:19 2024 kern.warn   kernel: [ 1608.099454] rtw_8822bu 1-1:1.0: timed out to flush queue 2
Wed Sep 18 22:01:19 2024 kern.warn   kernel: [ 1608.299433] rtw_8822bu 1-1:1.0: timed out to flush queue 3
Wed Sep 18 22:01:24 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: deauthenticated due to   local deauth request
Wed Sep 18 22:01:31 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: authenticated
Wed Sep 18 22:01:31 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: associated (aid 1)
Wed Sep 18 22:01:31 2024 kern.warn   kernel: [ 1620.549033] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:32 2024 kern.warn   kernel: [ 1621.548999] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:33 2024 kern.warn   kernel: [ 1622.548980] rtw_8822bu 1-1:1.0: failed to get tx report from   firmware
Wed Sep 18 22:01:37 2024 kern.warn   kernel: [ 1626.188894] rtw_8822bu 1-1:1.0: timed out to flush queue 3
Wed Sep 18 22:01:51 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: authenticated
Wed Sep 18 22:01:51 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 IEEE 802.11: associated (aid 1)
Wed Sep 18 22:01:52 2024 daemon.notice   hostapd: wlan0: AP-STA-CONNECTED 86:f8:c4:76:b7:27
Wed Sep 18 22:01:52 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 RADIUS: starting accounting session   2C54B67393F3E882
Wed Sep 18 22:01:52 2024 daemon.info   hostapd: wlan0: STA 86:f8:c4:76:b7:27 WPA: pairwise key handshake completed   (RSN)
Wed Sep 18 22:01:52 2024 daemon.notice   hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 86:f8:c4:76:b7:27
Wed Sep 18 22:01:52 2024 daemon.info   dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.178 86:f8:c4:76:b7:27
Wed Sep 18 22:01:52 2024 daemon.info   dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.178 86:f8:c4:76:b7:27
Wed Sep 18 22:02:06 2024 kern.err kernel:   [ 1654.934497] rtw_8822bu 1-1:1.0: error beacon valid
Wed Sep 18 22:02:06 2024 kern.err kernel:   [ 1654.935337] rtw_8822bu 1-1:1.0: failed to download drv rsvd page
Wed Sep 18 22:02:07 2024 kern.err kernel:   [ 1655.738028] rtw_8822bu 1-1:1.0: error beacon valid
Wed Sep 18 22:02:07 2024 kern.err kernel:   [ 1655.738961] rtw_8822bu 1-1:1.0: failed to download drv rsvd page
Wed Sep 18 22:02:07 2024 kern.err kernel:   [ 1656.069249] rtw_8822bu 1-1:1.0: error beacon valid
Wed Sep 18 22:02:07 2024 kern.err kernel:   [ 1656.070086] rtw_8822bu 1-1:1.0: failed to download drv rsvd page
---------
  ——————内核日志——————

[   76.316545] overlayfs:   upperdir is in-use as upperdir/workdir of another mount, accessing files from   both mounts will result in undefined behavior.
--
[   76.316564] overlayfs: workdir is in-use as   upperdir/workdir of another mount, accessing files from both mounts will   result in undefined behavior.
[   76.316574] overlayfs: lowerdir is in-use   as upperdir/workdir of another mount, accessing files from both mounts will   result in undefined behavior.
[   76.316580] overlayfs: lowerdir is in-use   as upperdir/workdir of another mount, accessing files from both mounts will   result in undefined behavior.
[   77.088177] IPv6: ADDRCONF(NETDEV_CHANGE):   wlan0: link becomes ready |   |   |   |   |   |   |   |   |  
[   77.088634] br-lan: port 2(wlan0) entered   blocking state |   |   |   |   |   |   |   |   |   |  
[   77.088660] br-lan: port 2(wlan0) entered   forwarding state |   |   |   |   |   |   |   |   |   |  
[   89.112891] rtw_8822bu 1-1:1.0: failed to   get tx report from firmware |   |   |   |   |   |   |   |  
[   90.112980] rtw_8822bu 1-1:1.0: failed to   get tx report from firmware |   |   |   |   |   |   |   |  
[   90.942860] rtw_8822bu 1-1:1.0: failed to   get tx report from firmware |   |   |   |   |   |   |   |  
[  184.370838] rtw_8822bu 1-1:1.0: timed out   to flush queue 3 |   |   |   |   |   |   |   |   |   |  
[  184.740836] rtw_8822bu 1-1:1.0: timed out   to flush queue 3 |   |   |   |   |   |   |   |   |   |  
[  209.499937] rtw_8822bu 1-1:1.0: failed to   get tx report from firmware |   |   |   |   |   |   |   |  
[  213.629755] rtw_8822bu 1-1:1.0: timed out   to flush queue 3 |   |   |   |   |   |   |   |   |   |  
[  656.170377] rtw_8822bu 1-1:1.0: error   beacon valid |   |   |   |   |   |   |   |   |   |   |  
[  656.171218] rtw_8822bu 1-1:1.0: failed to   download drv rsvd page |   |   |   |   |   |   |   |   |  
[  656.303000] rtw_8822bu 1-1:1.0: error   beacon valid |   |   |   |   |   |   |   |   |   |   |  
[  656.303842] rtw_8822bu 1-1:1.0: failed to   download drv rsvd page |   |   |   |   |   |   |   |   |  
[  657.877626] rtw_8822bu 1-1:1.0: error   beacon valid |   |   |   |   |   |   |   |   |   |   |  
[  657.878464] rtw_8822bu 1-1:1.0: failed to   download drv rsvd page |   |   |   |   |   |   |   |   |  
[  658.013628] rtw_8822bu 1-1:1.0: error   beacon valid |   |   |   |   |   |   |   |   |   |   |  
[  658.014466] rtw_8822bu 1-1:1.0: failed to   download drv rsvd page |   |   |   |   |   |   |   |   |  
[  658.148001] rtw_8822bu 1-1:1.0: error   beacon valid |   |   |   |   |   |   |   |   |   |   |  
---
出现这样的情况后，不能连接WIFI 。重启后，可以再次连接。
但在30分钟内，将会再次出现。

**（4）本人处置经过**

   1）如果刷2023年10月以前的固件，网络中没有【无线】选项
   2）网上查询，有说内核更新中自带rtw_8822bu驱动，且具有优先权。尝试网上说的balicklist禁用rtw_8822bu相关模块，禁用失败，任然启动。

5）希望能够给出解决此问题的建议，感谢！
     a、用2023年10月以前的固件，能取得【无线】选项的方法。
     b、或禁用rtw_8822bu加载的方法。
     c、又或者是其他能够解决AP不失去连接的方法

hscui, 2024-09-19:
这个我很早之前尝试时就发现与固件版本有关系，当时试过2022年中的每周更新：8月有问题，9月没问题，10月没问题，11月又有问题，我也试着查找过，似乎是内核问题。

zhukai168, 2024-09-19:
谢谢你的回答。后来是怎么解决这个问题的呢？还是放弃了吗？
现在也没有办法找到2022年的固件了，最晚的都是2023年3月份的。

hscui, 2024-09-19:
> 谢谢你的回答。后来是怎么解决这个问题的呢？还是放弃了吗？ 现在也没有办法找到2022年的固件了，最晚的都是2023年3月份的。

我当初是问了快半个月，没人回复，最后只能一个一个挨着试。所以要么自己试，要么就等大佬来回复。

zhukai168, 2024-09-25:
> > 谢谢你的回答。后来是怎么解决这个问题的呢？还是放弃了吗？现在也没有办法找到2022年的固定了，最晚的都是2023年3月份的。
> 
> 我一开始就是问了快半个月了，没人回复，最后只能一个个挨着试。所以要么自己试，要么就等大佬来回复。

能否告诉我哪一版固件，能显示[无线]而有没有掉线的问题啊?2023年9月以后的固件，有[无线]选项，但都会出现wifi掉线。
而2023年9月份以前的固件，均没有[无线]选项，也启用不了wifi ，没有wifi就失去了意义！盼给与指导，不甚感激！

zhukai168, 2024-09-25:
能否告诉我哪一版固件，能显示[无线]而有没有wifi掉线的问题啊？2023年9月以后的固件，有[无线]选项，但都会出现wifi掉线。
而2023年9月份以前的坚固，均没有[无线]选项，也启用不了wifi，没有wifi就失去了意义！盼给与指导，不甚感激！

hscui, 2024-09-25:
> 能否告诉我哪一版固件，能显示[无线]而有没有wifi掉线的问题啊？2023年9月以后的固件，有[无线]选项，但都会出现wifi掉线。 而2023年9月份以前的坚固，均没有[无线]选项，也启用不了wifi，没有wifi就失去了意义！盼给与指导，不甚感激！

我都没印象了，中间换过电脑，那个Comfast的USB后来给亲戚插他那台坏了WIFI的笔记本上了。不好意思。

closed 2024-10-10 by zhukai168