istoreos/istoreos #1823: Docker容器创建不出来

gh4zw, 2024-09-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Docker容器创建不出来，执行 Docker run 命令，提示“没有指定命令选项”，添加了命令选项后，又无法识别。同样的 Docker run 语句之前在群辉上是可以正常执行的。

### (2) 路由器型号和固件版本

A：星际蜗牛：Intel(R) Celeron(R) CPU J1900 @ 1.99GHz
      固件版本：iStoreOS 22.03.7 2024080210 / LuCI istoreos-22.03 branch git-24.207.28853-3b34de8

### (3) 详细日志和/或截图

A：[image]

rayangl, 2024-09-24:
有没有可能 不需要 run 你使用的命令是其他操作系统的

jjm2473, 2024-09-24:
eoch是啥命令啊？你确定不是echo？

gh4zw, 2024-09-24:
谢谢两位的回复，使我理顺了思路，找到自己忽略的地方了。
确实是界面表单填写有问题