istoreos/istoreos #1825: 希望尽快更新firewall4或者开发f4分支，越来越多插件需要了，官方版本都发布这么久了

PoemMistyMoon, 2024-09-23:
一直在用iStore，感觉首页界面很适合我，但是为什么 
  一直不更新openwrt版本和firewall4?
现在很多插件没有firewall4（nftables）的支持无法使用
如果担心firewall4稳定性问题，单独开发一个分支版本也可以，让想用的人自己去找这个分支来用，谢谢

jjm2473, 2024-09-24:
哪有越来越多插件需要fw4？大部分插件同时兼容fw3和fw4。
之所以没升级fw4，不是稳定性问题，而是兼容性问题，有不少插件只有fw3版本，例如部分游戏加速插件，简单网络控制等。
下个版本可能会升级到fw4，到时放弃不兼容的插件。

nobodysclown, 2024-09-25:
homeproxy 和 MihomoTproxy 都需要firewall4，用不上很难受

PoemMistyMoon, 2024-09-26:
> 虽然我是支持上fw4的，但homeproxy其实没你想象的好用 tproxy fw3也能用啊，装个iptables-mod-tproxy就行了
> 
> > 下个版本可能会升级到fw4
> 
> 下个版本不会是说6.6内核的吧？ 现在24.xx还没消息呢，不知道会不会变成25.xx 如果这样的话我也建议搞个fw4的放 https://fw.koolcenter.com/iStoreOS/alpha/

你可能回复错了，我需要fw4是因为想使用一些新功能和特性，你说的那两个代理程序我并没有使用也不是我回复的

bilibili2077, 2024-09-27:
一直在等firewall4，不知道啥时候能用上！

Jetzow527, 2024-09-27:
我现在还希望不要升级Firewall4。因为这样做会导致Tailscale使用起来非常困难。

bilibili2077, 2024-10-04:
+2

iErics, 2024-10-13:
+3

cjsycr, 2024-10-30:
+4

gzxiexl, 2024-11-09:
> 我现在还希望不要升级Firewall4。因为这样做会导致Tailscale使用起来非常困难。

PVE 单独建个LXC的tailscale 设置静态路由到tailscale 个人觉得更简洁

closed 2025-05-06 by jjm2473

xieyouming-Web, 2025-05-19:
> 一直在用iStore，感觉首页界面很适合我，但是为什么 一直不更新openwrt版本和firewall4？ 现在很多插件没有firewall4（nftables）的支持无法使用 如果担心firewall4稳定性问题，单独开发一个分支版本也可以，让想用的人自己路由这个分支来用，谢谢

现在24.10版本已经支持firewall4了