istoreos/istoreos #1829: Tailscale版本过于古老，能安排升级吗？

Jetzow527, 2024-09-25:
目前最新版本的iStoreOS里提供的版本仍然是1.32.3，这个版本已经过于古老，而且存在一定的安全漏洞，请问开发组有没有进行更新的计划？

bilibili2077, 2024-09-27:
里面的插件都没那么快更新的，早则一个月之内，晚则一年半载，所以自己手动安装最好！

Jetzow527, 2024-09-27:
> 里面的插件都没那么快更新的，早则一个月之内，晚则一年半载，所以自己手动安装最好！

是的，所以我手动更新了。
至少在1.56版本及之后，Tailscale已经给出了update参数，虽然还是[beta]，但目前用着没有什么不良反应。

yb9210, 2024-10-28:
https://github.com/CH3NGYZ/tailscale-openwrt  可以用这个，自动更新的

Jetzow527, 2024-10-28:
> https://github.com/CH3NGYZ/tailscale-openwrt 可以用这个，自动更新的

这个看起来好麻烦的样子。Tailscale官方版本现在也可以在线升级了，所以问题不算太大。

yb9210, 2024-12-09:
> > https://github.com/CH3NGYZ/tailscale-openwrt 可以用这个，自动更新的
> 
> 这个看起来好麻烦的样子。Tailscale官方版本现在也可以在线升级了，所以问题不算太大。

可以发我链接么，我想看看官方的怎么安装到istoreos

githubzxk, 2024-12-29:
根据上面的方法翻译成小白看得懂的方法。。

在 [pkgs.tailscale.com/stable/#static](https://pkgs.tailscale.com/stable/#static)中找到对应版本的包，比我我的x86_64软路由istoreos系统对应的包为amd64: tailscale_***_amd64.tgz；
下载下来这个tgz文件；
用ssh软件连接istoreos（最好用带文件管理的，比如FinalShell）；
输入停用tailscale命令： service tailscale stop
打开目录/usr/sbin；
用解压软件打开tgz，将里面的tailscale和tailscaled文件拷贝到/usr/sbin；
输入启用tailscale命令： service tailscale start
输入启用查询版本命令： tailscale version
输入更新命令：tailscale update
之后就可以使用tailscale update命令更新tailscale了

TKMYing, 2025-03-19:
+1 但是楼下的朋友给的方案也很好！