istoreos/istoreos #1831: smb配置被自动恢复

captain9293, 2024-09-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；
系统重启后，smb.conf总是被莫名奇妙地恢复

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用smb共享了两个目录，将其中一个共享删除，重启istoreos，被删除的共享目录又被恢复了，尝试直接修改smb.conf文件，问题依然在，重启istoreos后，smb.conf里面被删除的那段配置被恢复了，检查文件修改时间，变成了istoreos启动的时间

### (2) 路由器型号和固件版本

A：x86系统，固件版本号为iStoreOS 22.03.7 2024080210

### (3) 详细日志和/或截图

A：

jjm2473, 2024-09-26:
你是不是使用了“统一文件共享”？

jjm2473, 2024-09-26:
直接修改smb.conf肯定是没用的，因为这个文件是生成的，每次启动samba都会重新生成

xiaomeng9597, 2024-09-26:
> 直接修改smb.conf肯定是没用的，因为这个文件是生成的，每次启动samba都会重新生成

处理一下拉取请求

captain9293, 2024-09-26:
> > 直接修改smb.conf肯定是没用的，因为这个文件是生成的，每次启动samba都会重新生成
> 
> 处理一下拉取请求

怎么处理？

ymsq, 2024-10-06:
> 你是不是使用了“统一文件共享”？

我遇到了相似的问题，在统一文件共享授予写入权限后依然没有写入权限，webdva也是。目前使用三方插件GoWebDav代替，写入正常，但是我所使用的移动设备不支持WebDav，希望修复

b0070, 2025-03-11:
同样问题，一直自动恢复