istoreos/istoreos #1844: 内网测速只有百兆，是怎么回事

walkerfirst, 2024-10-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

dell 5070 这个机器物理机和群晖虚拟机安装istore os，网速只有100M。
已排除网线网卡的问题，同一套网络安装windows和其他版本的openwrt就没有问题。

内网测试：物理机 下载正常，上传只有百兆。群晖虚拟机 下载上传都是百兆。

A：

### (2) 路由器型号和固件版本

A：Dell 5070 瘦客户端机  
[istoreos-22.03.7-2024080210-x86-64-squashfs-combined-efi](https://fw0.koolcenter.com/iStoreOS/x86_64_efi/istoreos-22.03.7-2024080210-x86-64-squashfs-combined-efi 和之前的版本都尝试过，问题依旧。

### (3) 详细日志和/或截图

A：

walkerfirst, 2024-10-08:
[image]
[image]

jjm2473, 2024-10-22:
你的内网测速都是300和500，你确定不是在用wifi？