istoreos/istoreos #1848: Radxa E20C重启死机

llhdxnq, 2024-10-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在后台点击重启按钮，不会自动开机。 LED灯常亮，必须要重新插拔电源才可以。

### (2) 路由器型号和固件版本

A：iStoreOS 22.03.7 2024080210

### (3) 详细日志和/或截图

A：第一张截图是设备重启前日志，第二张图是设备重启后日志。
[image]
[image]
[image]

jjm2473, 2024-10-22:
没发现这个问题，建议咨询radxa官方或者换tf卡试试

llhdxnq, 2024-11-14:
> 没发现这个问题，建议咨询radxa官方或者换tf卡试试

已解决，确实是TF卡的问题