istoreos/istoreos #1853: 自编译后没有quickstart命令

sclsyin, 2024-10-17:
使用活跃分支 istoreos-22.03 编译后，虚拟机安装好，发现没有quickstart命令，请问要在哪选择才有？

jjm2473, 2024-10-22:
如果要跟istoreos官方的固件一样，可以使用istoreos官方固件的编译配置文件。打开 https://fw.koolcenter.com/iStoreOS/ ，找到自己的机型，进去目录，下载里面的`feeds.conf`和`config.seed`。将`feeds.conf`放在istoreos仓库根目录然后执行`./scripts/feeds update -a && ./scripts/feeds install -a`，没错误的话继续。将`config.seed`放在istoreos仓库根目录，并重命名成`.config`，然后执行`make defconfig`，没错误的话执行`make V=s -j4`构建就行了。

此外还可以参考这个项目 https://github.com/Siriling/istoreos-actions

xiaomeng9597, 2024-11-09:
> 如果要跟istoreos官方的固件一样，可以使用istoreos官方固件的编译配置文件。打开 https://fw.koolcenter.com/iStoreOS/ ，找到自己的机型，进去目录，下载里面的`feeds.conf`和`config.seed`。将`feeds.conf`放在istoreos仓库根目录然后执行`./scripts/feeds update -a && [...]

你推荐的这个仓库，11月份他编译的一样没有luci-app-quickstart，luci-i18n-quickstart-zh-cn，luci-app-store，11月份以前的就有，不可能是环境问题，第三方编译istoreos已经寄了

xiaomeng9597, 2024-11-09:
> 使用活跃分支 istoreos-22.03 编译后，虚拟机安装好，发现没有quickstart命令，请问要在哪选择才有？

原来自带的，还有官方固件网站里面的.config配置文件里面也有luci-app-quickstart，luci-i18n-quickstart-zh-cn，quickstart这些包名，即便添加上面几个插件包名编译出来的固件都没有编译这几个插件进去，只有一个quickstart依赖编译进去了，luci-app-quickstart，luci-i18n-quickstart-zh-cn这两个插件是没有的编译进去的，连istore应用商店也编译不进去了，加了luci-app-store包名都不行，建议放弃吧

jjm2473, 2025-07-15:
@karlhu85 那你进软件包看看是不是已经有 luci-i18n-quickstart-zh-cn，如果没有，那就是编译配置问题。

closed 2025-07-15 by jjm2473