istoreos/istoreos #1860: 无法识别万兆网卡

cuteyboys, 2024-10-21:
具体问题

A：我是用硬酷R2 Max的万兆机器 万兆网卡为AQC113C-B1-C 安装了固件后 可能由于驱动问题 无法识别万兆网卡 该怎么解决

路由器型号和固件版本

A：固件版本为 2023052616

jjm2473, 2024-10-22:
用最新固件来测试，然后执行`lspci -nnk`把输出贴出来

cuteyboys, 2024-10-22:
[image]
固件版本：22.03.7-2024080210

cuteyboys, 2024-10-23:
> 用最新固件来测试，然后执行`lspci -nnk`把输出贴出来

[image]
固件版本：22.03.7-2024080210

jjm2473, 2024-10-25:
今天会发固件，集成了驱动，需要自己测试下有没有问题。

cuteyboys, 2024-10-25:
> 今天会发固件，集成了驱动，需要自己测试下有没有问题。

老哥 更新固件现在还没有发出吗

jjm2473, 2024-10-27:
@cuteyboys 你先到这里下载吧，不过安装以后先别用在线更新，不然会降级 https://sourceforge.net/projects/istoreos/files/iStoreOS/

jjm2473, 2024-10-29:
检查更新时如果显示202408xx版本，就别更新，至少要202410以后的才行。这周应该会发上线的

closed 2024-10-29 by jjm2473